micro-manager/micro-manager #211: mm2: Splash screen: Should disable 'Delete' for 'Create new profile'

marktsuchida, 2016-02-19:


ChrisWeisiger, 2016-02-19:
I don't understand this one. When you select "Create new profile", a dialog pops up for the new profile's name. No matter what you do from that dialog, the primary dialog doesn't let you stay on the "Create new profile" option (it either goes to the newly-created profile, or to the default profile), so there should be no way to try to delete the "Create new profile" option.

What am I missing?

marktsuchida, 2016-02-20:
Sorry for the insufficient description. I figured out now what had happened.

1. Create a profile named "TEST".
2. Delete the "TEST" profile.
3. Immediately try creating a profile named "TEST" again.
4. This fails (bug).
5. At this point, the popup has "Create new profile" selected (bug).
6. And you can actually delete the "Create new profile" item (bug).
7. Quit and restart MM.
8. The "TEST" profile still show up (bug).

ChrisWeisiger, 2016-02-22:
Thanks for the more detailed repro steps. This is now fixed; there was a bug with the "delete profile" logic and everything else cascaded from that.

closed 2016-02-22 by ChrisWeisiger