micro-manager/micro-manager #216: mm2: Line selection tools doesn't work on images

marktsuchida, 2016-02-21:
In particular, the line used for Line Profile cannot be adjusted.

ChrisWeisiger, 2016-02-21:
That's odd; I swear I used the line profile to test that the histograms back when I was updating them to work with RGB images. Anything special you're doing?

marktsuchida, 2016-02-21:
Turns out the line selection tool doesn't work when there is no pixel size configuration (testable by deleting all pixel size presets from demo config).

ChrisWeisiger, 2016-02-22:
This appears to be a bug in ImageJ's ROI logic. If I have no pixel size config, then I'm unable to draw line ROIs; they immediately disappear as soon as I let go of the mouse. Other ROIs are possible, but of course the Line Profile window doesn't use those. 

The Line Profile window is somehow able to cause a line profile to appear on the display anyway, but ImageJ believes there's no ROI (getROI() returns null), causing the Line Profile to revert to its default line.

It's unclear to me what exactly is going on. It's possible that newer versions of ImageJ have fixed this problem; I haven't tested them.

ChrisWeisiger, 2016-02-22:
One option would be to provide dummy pixel size information if no pixel config is available. For example, we can set a pixel size of 1.0e-20µm (.01 yoctometers), and the line profile will behave itself properly. It's only when we have a non-positive value that we run into problems.

ChrisWeisiger, 2016-02-22:
After discussion with Mark, we now set the pixel size unit to pixels instead of microns, which allows us to have a sensible "size" (of 1 pixel per pixel), which allows the line ROI to work. This is now fixed.

closed 2016-02-22 by ChrisWeisiger