micro-manager/micro-manager #218: mm2: Code subscribing to DisplayDestroyedEvent assumes DefaultDisplayWindow

nicost, 2016-02-25:
Trying to be a good citizen and generating a DisplayDestroyedEvent when the ClearVolume Display Window exits generates several:
```SEVERE: Could not dispatch event: ```
by the eventbus.  One of these is in org.micromanager.internal.navigation.ClickToMoveManager.onDisplayDestroyed, which casts the DisplayWindow to a DefaultDisplayWindow without checking if that cast is possible, hence throws an exception.  Is this a bug, or is there the assumption that such events should only be generated by the DefaultDisplayWindow and never by other DisplayWindows? 

I see the same 
``SEVERE: Could not dispatch event:```
coming from "org.micromanager.display.internal.inspector.InspectorFrame", but could not figure out the reason in that case.

ChrisWeisiger, 2016-02-25:
Per the documentation for DisplayDestroyedEvent: "This event signifies that a DisplayWindow has been closed (that its forceClosed() method was called). It is not called for DataViewers." I admit this could be clearer, but it is not expected that a DataViewer publish one of these events when it goes away. All you should need to do is call `DisplayManager.removeViewer()` (assuming that you had called `DisplayManager.addViewer()` previously). 

I've added a note to each of the *Event classes in the display API to indicate if third-party code is expected to produce that event.

nicost, 2016-02-25:
Some confusion here, my code (https://github.com/nicost/MMClearVolumePlugin) is still implementing a DisplayWindow, not a a DataViewer, since otherwise the histograms do not show up (issue we talked about a long time ago and that never was resolved).  So, the description of that function suggests that I do have to call it.  

Also, I assume that it is supposed to be possible for 3d party code to implement DisplayWindows and to call these events.  If so, that will surely exceptions.

ChrisWeisiger, 2016-02-25:
No, it is not intended for third-party code to implement DisplayWindow, nor do I anticipate it ever being intended. DataViewer is explicitly meant to be the third-party display interface. DisplayWindow is µManager internal code only. What we need to do here is fix whatever issue is keeping histograms from working with your plugin so you can implement DataViewer instead of DisplayWindow. I thought that had been resolved; can you refresh my memory on what happens when you try to use DataViewer?

nicost, 2016-02-25:
Just reverted from DisplayWindow to DataViewer and all is fine, so please disregard this thread (since it seems to be based on confusion caused by a murky past;).  Only thing that would be nice is if there were a way to programmatically prohibit 3rdparty code to inherit from DisplayWindow (but I would not know how to do that).

ChrisWeisiger, 2016-02-26:
Glad to hear that the DataViewer approach is working!

Unfortunately, I don't think it's possible to have a public interface that people cannot implement, since the Java compiler has no way to tell the difference between µManager's internal implementation (which must be allowed) and some random third-party implementation (which must be blocked). The DisplayWindow documentation does note at the top that third-party code is not intended to implement it, and points them to DataViewer instead. I think that's about the best we can do right now.

closed 2016-02-26 by ChrisWeisiger

marktsuchida, 2016-02-26:
This is a common situation in Java (as you might imagine), and it is considered good practice to clearly distinguished in the documentation whether a given class or interface is intended to be subclassed by client code (i.e. it is a service provider interface) or not (i.e. it is a normal API). As Chris says, there's nothing more that can be done.