micro-manager/micro-manager #223: mm2: Multi-Dimensional Acquisition Window looks strange on Windows

nicost, 2016-02-26:
Especially in the Z-stacks section, there is no space between the start, end, and set size boxes.  This is on Windows 7, using Java 1.8.0_73.  Same thing with the nightly build and Java 6.

[image]

ChrisWeisiger, 2016-02-26:
This should be significantly improved now. Thoughts?

marktsuchida, 2016-02-27:
I have created a patch that fixes several remaining issues (incorrect resizing behavior, lack of scrolling in comments area, inconsistent horizontal positioning of upper panel items, inconsistent capitalization and label alignment, etc.).
Yes, the channel table resizes now!
I still need to test on Windows before merging, just letting you know so there's no duplication of effort.

marktsuchida, 2016-02-28:
Created PR #225 - tested on Windows and OS X.

closed 2016-03-03 by ChrisWeisiger