micro-manager/micro-manager #226: mm2: Live display stops updating after a while

nicost, 2016-03-04:
Start live mode (Andor Zyla, ~10ms exposure time).  After a couple of minutes, the display stops updating.  Stopping and starting live mode again does not restore the display.  Closing the window and pressing live again restarts the live display, suggesting this is an issue in the display rather than the hardware.

ChrisWeisiger, 2016-03-23:
Sorry, refresh my memory: is this still an issue? I know I've fixed some live stuff in the past few weeks, but I forget if this was one of them.

nicost, 2016-03-23:
I think that you fixed this partly on 3/4/2016 
(https://github.com/micro-manager/micro-manager/commit/570f30fccc3764bfc3e8d841756399db9571d0c3), 
and then more fully on 3/7/2016 
(https://github.com/micro-manager/micro-manager/commit/b4e4a6c5836e9544976dbc45b8981622dc09ddde). 
Those appear to have resolved the issue.

ChrisWeisiger, 2016-03-23:
Great, thanks!

closed 2016-03-23 by ChrisWeisiger