micro-manager/micro-manager #229: Saved files are locked and can not be touched by other applications untill MM quits

nicost, 2016-03-17:
Snap an image, click on the disk icon to save the image. Close the image.  Locate the file o0r folder on disk and try to move it to another location.  On Windows (I did not test recently on Mac) this can not be done because mm locks the file.  Same problem exists in 1.4.

marktsuchida, 2016-03-17:
This problem has been around for a while and would be nice to fix if possible. This wouldn't be easily detected on Unix (including Mac) because you can move/delete open files anyway.

We use `java.nio` file channels to write image files - perhaps we are not closing the channel (or whatever related object that needs closing) under some or all conditions?

ChrisWeisiger, 2016-03-23:
This is now fixed; when the last display for a "managed" Datastore is closed, Datastore.close() is called, and that in turn will eventually result in the StorageMultipageTiff cleaning up its file descriptors. This does make it impossible to read anything from the Datastore, so anyone who's relying on it in third-party code should be listening for the DatastoreClosingEvent. That's not super-ideal, but I think it's preferable to µManager keeping those file descriptors open indefinitely on the off-chance that someone else is using them.

ChrisWeisiger, 2016-03-23:
By the way, you can see a list of open file descriptors on Mac by using the Activity Monitor program, selecting the running program, and then clicking on the "info" button (script 'i' in a circle). On Linux, you can interrogate /proc/<pid> to see open file descriptors.

closed 2016-03-23 by ChrisWeisiger

marktsuchida, 2016-03-24:
Are we relying on garbage collection and finalizers to ultimately close the `FileChannel`s? I'm having trouble figuring out how `DefaultDatastore.close()` results in `FileChannel.close()`. (Where is `MultipageTiffReader.close()` called?)

If we are relying on finalizers for now it's probably not urgent to do otherwise (it might require risky refactoring), but it will be good to explicitly close all files we open (to avoid resource leaks that are platform-, system-, and phase-of-the-moon-dependent). (But we need to resign ourselves to relying on finalizers when it comes to RAM storage since there is no proper way to explicitly release a `java.nio` buffer.)

ChrisWeisiger, 2016-03-24:
`DefaultDatastore.close()` calls `StorageMultipageTiff.close()`, which either directly calls `MultipageTiffReader.close()` (for read-only datastores) or goes by way of the `FileSet` (for read/write datastores). GC/finalizers are not involved; it's just a direct call to `FileChannel.close()` (and `RandomAccessFile.close()`).

marktsuchida, 2016-03-24:
And of course I was simply looking at the wrong version of the code. Thanks for the explanation.