micro-manager/micro-manager #230: mm2: standard deviation of images is no longer reported.

nicost, 2016-03-17:
In 1.4 the standard deviation is reported together with min, max, and mean.  This is often a useful number, especially when doing quantitative work.  It would be nice if 2.0 would also report the std. deviation.

ChrisWeisiger, 2016-03-18:
I note that 1.4 only did standard deviation for single-channel datasets (though it's possible this is simply an oversight as 1.4 uses separate code paths for single- and multi-channel histograms), and uses ImageJ's algorithm to calculate the variance, which appears to be a fairly naive implementation that Wikipedia says is subject to potential numerical instability (though I lack the mathematical expertise to evaluate if that claim is relevant in our case). It also requires iterating over every value in the histogram.

I tried implementing the Knuth algorithm described on Wikipedia, which is a single-pass algorithm, so it can be calculated while we "touch" each pixel. It does however require performing an extra division operation for each pixel. It increases the runtime of the histogram/stats calculation for a 512x512 image from 1.3ms to 1.8ms; for a 1k x 1k image it goes from 4.8ms to 7.2ms. For reference, the "ImageJ algorithm" on a 1k x 1k image takes our cost to 5.2ms -- so at least it's cheaper, though again we need to check on that numerical instability thing to see if it's a problem.

Considering that stats calculation is _the_ major limitation on our framerate, I'm hesitant to add this to every single image we touch. Would it be acceptable to have a separate step you have to perform to get the standard deviation (i.e. it is not automatically calculated for you)?

nicost, 2016-03-19:
> I note that 1.4 only did standard deviation for single-channel datasets
>

I do think that SD is mainly of interest for single channel datasets.  
It is an important parameter when setting up an experiment, and/or when 
determining characteristics of a camera, so is very useful feedback in 
single channel data, less so in multi-channel acquisitions.
>
> (though it's possible this is simply an oversight as 1.4 uses separate 
> code paths for single- and multi-channel histograms),
>
The code for SD was added by Photometrics.  All serious imaging 
applications provide this number, and you can be sure that camera 
company people will note its absence.
> I tried implementing the Knuth algorithm described on Wikipedia, which
> is a single-pass algorithm, so it can be calculated while we "touch"
> each pixel. It does however require performing an extra [...]
So, the ImageJ method adds 0.4 ms for the 1k x 1k image, or ~8%. Seems 
worth it to me.  I guess that numerical instability could result from 
the sum of squared errors overflowing.  I have never noted that and 
would assume that this is not going to be a problem for the foreseeable 
future (i.e. with the image sizes we work with).
>
> Considering that stats calculation is /the/ major limitation on our 
> framerate, I'm hesitant to add this to every single image we touch. 
> Would it be acceptable to have a separate step you have to perform to 
> get the standard deviation (i.e. it is not automatically calculated 
> for you)?
>
Then how do I get that to work on a live image?  Having to click 
everytime I want that number updated is not really a solution.

ChrisWeisiger, 2016-03-24:
The above commit adds a gear menu option to turn standard deviation calculation on if the user so desires (and that setting will be remembered across sessions). I believe this should be a satisfactory solution.

closed 2016-03-24 by ChrisWeisiger