micro-manager/micro-manager #232: mm2: Most device properties are not retrievable in the metadata

nicost, 2016-03-18:
Just discovered that the metadata of images saved in 2.0 do not contain the metadata about the device properties (or at least, those are not shown when opening images in either 1.4 or 2.0).  It is absolutely essential that those properties can be retrieved, and I will have to stop using 2.0 until those properties are included in the metadata again.

ChrisWeisiger, 2016-03-18:
Just tested this: ran a five-timepoint DemoCamera acquisition, saved it as a stack file, and re-opened it. All device properties are present. Can you expound upon your test process? You did remember to click the "Show Unchanging Properties" checkbox, right?

nicost, 2016-03-18:
Start MM with demo config.  Snap an image.  Look at metadata and find 
that there is no information about properties of any of the devices. 
Saving and re-opening does not make that information appear.

Add an image to an album.  Check the metadata.  No device properties to 
be found (even the exposure time is lost).

That just made it impossible to check the experiment I was doing 
yesterday...

B.t.w., 1.4 behaves as expected (i.e. all properties are shown when 
snapping an image).

ChrisWeisiger, 2016-03-18:
This appears to be specific to the Snap/Live window (and images generated via the Snap/Live window and then subsequently saved). Can you confirm that MDA works as expected? Not having metadata in the Snap/Live view is still a bug of course. Or was; it's now fixed.

Closing; feel free to re-open if you encounter problems with image metadata.

closed 2016-03-18 by ChrisWeisiger