micro-manager/micro-manager #238: mm2: DisplayWindow window size off by a few pixels

marktsuchida, 2016-03-24:
Upon opening a new display window, and upon clicking on the zoom in/out buttons, the window is supposed to resize to match the image exactly, as long as there is enough space on the screen. At the moment the window appears to always be just a few pixels (might be 1 pixel, hard to tell) smaller than it should be, resulting in the purple ImageJ navigator to show up in the top-left corner.

(Not to be confused with #220.)

ChrisWeisiger, 2016-04-04:
This has been _mostly_ fixed by the above commit, but you can still occasionally get some incorrectly-sized image windows when adding scrollbars during an MDA. Just do a multi-time/Z/channel acquisition and sometimes when a new axis is started, the corresponding scrollbar will steal window space from the image display instead of the image window growing. There's probably a race condition somewhere in the code.

nicost, 2020-04-04:
This can still happen on occasion (04-04-2020).  Hard to reproduce, and not high priority.