micro-manager/micro-manager #240: mm2 : inconsistency when processImage() does not call context.outputImage(image) each time

hadim, 2016-04-04:
When using Processor (see the [FrameAverager](https://github.com/hadim/FrameAverager) plugin for example), `processImage()` does not outputs an image (with `context.outputImage(image)`) at each call.

So for example when imaging 100 frames and outputing only 10 images, the resulting display window contains 100 frames (empty frames are filled with existing frames) but the OME XML contains only 10 planes and so do not open properly in Fiji (SCIFIO and Bioformats).

What do you suggest ?

ChrisWeisiger, 2016-04-04:
You need to "compact" the coordinate set. For example, if you are outputting images along the Time axis, then the first image sent to `outputImage()` must have a time coordinate of 0, the second must have a time coordinate of 1, etc. You can do that using `Image.copyAtCoords()` to generate new `Image` objects with the same pixel data and metadata, but different coordinates.

On a side note, your processor may also want to update the `intendedDimensions` field of the `SummaryMetadata` in `processSummaryMetadata`. I don't believe at this time that we actually rely on the field to be accurate, but it would be polite. :)

I hope this helps!

hadim, 2016-04-04:
That is exactly what I needed. Thank you @ChrisWeisiger !

closed 2016-04-04 by hadim