micro-manager/micro-manager #243: mm2 : exception at the end of MDA with DemoConfig

hadim, 2016-04-05:
See the log : 

```
2016-04-04T21:39:15.757200 tid139708695136000 [IFO,App] 
                                              [       ] java.lang.RuntimeException: java.lang.Exception: Property value "10,0000" contains reserved characters in Thread[AcquisitionEngine2010 Thread (Clojure),6,main]
                                              [       ]   at clojure.lang.Util.runtimeException(Util.java:165)
                                              [       ]   at clojure.lang.Reflector.invokeInstanceMethod(Reflector.java:35)
                                              [       ]   at org.micromanager.acq_engine$set_property$attempt__813__auto____896.invoke(acq_engine.clj:260)
                                              [       ]   at org.micromanager.acq_engine$set_property$fn__900.invoke(acq_engine.clj:260)
                                              [       ]   at org.micromanager.acq_engine$set_property.invoke(acq_engine.clj:260)
                                              [       ]   at clojure.core$map$fn__3811.invoke(core.clj:2432)
                                              [       ]   at clojure.lang.LazySeq.sval(LazySeq.java:42)
                                              [       ]   at clojure.lang.LazySeq.seq(LazySeq.java:60)
                                              [       ]   at clojure.lang.RT.seq(RT.java:466)
                                              [       ]   at clojure.core$seq.invoke(core.clj:133)
                                              [       ]   at clojure.core$dorun.invoke(core.clj:2723)
                                              [       ]   at org.micromanager.acq_engine$return_config.invoke(acq_engine.clj:533)
                                              [       ]   at org.micromanager.acq_engine$cleanup$fn__1194.invoke(acq_engine.clj:694)
                                              [       ]   at org.micromanager.acq_engine$cleanup.invoke(acq_engine.clj:686)
                                              [       ]   at org.micromanager.acq_engine$run_acquisition.invoke(acq_engine.clj:793)
                                              [       ]   at org.micromanager.acq_engine$run$fn__1414.invoke(acq_engine.clj:964)
                                              [       ]   at clojure.lang.AFn.run(AFn.java:24)
                                              [       ]   at java.lang.Thread.run(Thread.java:701)
                                              [       ] Caused by: java.lang.Exception: Property value "10,0000" contains reserved characters
                                              [       ]   at mmcorej.MMCoreJJNI.CMMCore_setProperty__SWIG_0(Native Method)
                                              [       ]   at mmcorej.CMMCore.setProperty(CMMCore.java:450)
                                              [       ]   at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
                                              [       ]   at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
                                              [       ]   at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
                                              [       ]   at java.lang.reflect.Method.invoke(Method.java:622)
                                              [       ]   at clojure.lang.Reflector.invokeMatchingMethod(Reflector.java:92)
                                              [       ]   at clojure.lang.Reflector.invokeInstanceMethod(Reflector.java:30)
                                              [       ]   at org.micromanager.acq_engine$set_property$attempt__813__auto____896.invoke(acq_engine.clj:260)
                                              [       ]   at org.micromanager.acq_engine$set_property$fn__900.invoke(acq_engine.clj:260)
                                              [       ]   at org.micromanager.acq_engine$set_property.invoke(acq_engine.clj:260)
                                              [       ]   at clojure.core$map$fn__3811.invoke(core.clj:2432)
                                              [       ]   at clojure.lang.LazySeq.sval(LazySeq.java:42)
                                              [       ]   at clojure.lang.LazySeq.seq(LazySeq.java:60)
                                              [       ]   at clojure.lang.RT.seq(RT.java:466)
                                              [       ]   at clojure.core$seq.invoke(core.clj:133)
                                              [       ]   at clojure.core$dorun.invoke(core.clj:2723)
                                              [       ]   at org.micromanager.acq_engine$return_config.invoke(acq_engine.clj:533)
                                              [       ]   at org.micromanager.acq_engine$cleanup$fn__1194.invoke(acq_engine.clj:694)
                                              [       ]   at org.micromanager.acq_engine$cleanup.invoke(acq_engine.clj:686)
                                              [       ]   at org.micromanager.acq_engine$run_acquisition.invoke(acq_engine.clj:793)
                                              [       ]   at org.micromanager.acq_engine$run$fn__1414.invoke(acq_engine.clj:964)
                                              [       ]   at clojure.lang.AFn.run(AFn.java:24)
                                              [       ]   at java.lang.Thread.run(Thread.java:701)
```

My feeling says that it's a localization bug in acquisition engine (that's why you didn't see it before probably).

I don't like having to tweak my local settings to fix this. A proper fix in the MM code would be cleaner in my opinion.

I'd like to help on this but I think I am allergic to Clojure :-D

ChrisWeisiger, 2016-04-07:
Sorry for not getting back to you on these more recently; I've been out sick for the past few days. For reference, what are your locale settings? Thanks!

hadim, 2016-04-08:
A mix between `fr_FR` and `en_US`.

```sh
$ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=fr_FR.UTF-8
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=fr_FR.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=fr_FR.UTF-8
LC_NAME=fr_FR.UTF-8
LC_ADDRESS=fr_FR.UTF-8
LC_TELEPHONE=fr_FR.UTF-8
LC_MEASUREMENT=fr_FR.UTF-8
LC_IDENTIFICATION=fr_FR.UTF-8
LC_ALL=
```

hadim, 2016-04-09:
`print(Locale.getDefault());` returns `en_US` in the script window of MM. Not sure it's related to locale config anymore...

hadim, 2016-04-09:
Note that the error only appears for the first MDA. Next MDA (without restarting MM) doesn't have the error...

ChrisWeisiger, 2016-04-09:
Interesting. What happens if you load µManager, go into the Device Property Browser, find any property that's set to 10000, change it, change it back, and then run MDA? How about if you hit the "Refresh" button in the main window before running the first MDA?

hadim, 2016-04-09:
No property is set to 10,000 and the "Refresh" button trick doesn't work neither.

hadim, 2016-04-09:
I just realized it's not even `10,000` in the log but `10,0000`. So weird...

marktsuchida, 2016-04-12:
"10,0000" is probably ten in European notation (comma as decimal point). Normally (under, say, fr locale) property values should be converted correctly to and from the localized representation, but maybe that is not working -- possibly due to the mixed locale settings, though I'm not sure. I would try forcing the locale to either French or English and see if it still happens.

hadim, 2016-04-13:
I still have the same issue launching MM with the following java option `-Duser.country=US -Duser.language=en `.

But I am not sure whether this option is the correct one to change the locale setting...

marktsuchida, 2020-08-28:
If this issue has not fixed itself in the last 4 years, I think we'll need full instructions to reproduce (starting with OS name and version so we can test in a similar virtual machine). Please reopen or file a new issue if necessary.

closed 2020-08-28 by marktsuchida