micro-manager/micro-manager #246: mm2 : Synchronize Windows doesn't work with  Live View and Album window

hadim, 2016-04-05:
While it work with Live View and a normal IJ window (A duplicate of Album window for example.

ChrisWeisiger, 2016-04-07:
To be clear, you're talking about Fiji's synchronization system? µManager has its own built-in image linking system, described here:

https://micro-manager.org/wiki/Version_2.0_Users_Guide#Linking_Displays

However, there was a bug in the Album code that caused it to not have a valid channel name, which meant that you couldn't use that linking system with the Album. I've fixed that in 3ffcbf450236612f356f9328ae6a580263469383.

hadim, 2016-04-07:
Yes exactly. 

And the thing is I often synchronize a IJ window with a MM window so Linking Displays feature won't be very helpful here...

ChrisWeisiger, 2016-04-08:
Okay, thanks for the clarification. I took a look at the Synchronize Windows logic, and unfortunately I can state with confidence that it categorically cannot possibly work with µManager image windows. The way it is written assumes that the GUI component that is the parent of the ImageCanvas is itself an ImageWindow:

`Calibration cal = ((ImageWindow)ic.getParent()).getImagePlus().getCalibration();`
(taken from the `getMatchingCoords()` method of ij.frame.SyncWindows)

This is not true for µManager 2.0's display windows, which are considerably more complicated than ImageJ windows (In fact, there is no `ImageWindow` anywhere in the GUI hierarchy above the `ImageCanvas`). Consequently, any time the Synchronize Windows system attempts to access the ImagePlus for a µManager window, it will attempt to cast a `JPanel` to `ImageWindow`, which fails, throwing an uncaught exception.

I suggest you file a bug with the creators of the Synchronize Windows plugin for ImageJ. They should be using `ImageCanvas.getImagePlus()` rather than their current system. I believe that should let them access the information they need without any brittle and failure-prone typecasts. I can't promise that fixing this will allow µManager windows to work with the Synchronize Windows feature, but it's certainly a necessary precondition. Note that they have casts to `ImageWindow` in many locations in their code; it's not just the one method I mentioned that will need to be fixed.

closed 2016-04-08 by ChrisWeisiger

hadim, 2016-04-08:
Thanks for the clarification. I will have a look at the code of "Synchronize Windows"