micro-manager/micro-manager #247: mm2 : add additional columns in the Channels section in the MDA window

hadim, 2016-04-05:
What do you think about adding additional columns in the Channels section in the MDA window ?

Each column could correspond to a Group and a different Preset could be loaded for different channel.

The use-case is for example when imaging in DIC and TIRF at the same time, I want to set the Camera Gain to a different value for both channel.

I know I can do that with a script and using Runnable (does it still exist in MM2 ?) but an additional column would help :-)

What do you think ?

PS : if I don't have to hack the acquisition engine in Clojure I can have a look and try to make a PR

ChrisWeisiger, 2016-04-05:
Unfortunately you would need to modify the acquisition engine to do this, as it only expects to deal with one channel group in any given acquisition.

Runnables still exist; check the AcquisitionManager (`Studio.acquisitions()`).

hadim, 2016-04-05:
Ok so it's probably better putting the camera gain in my channel group then. 

I close.

closed 2016-04-05 by hadim