micro-manager/micro-manager #248: mm2 : Enable / Disable checkbox for Processors does not work

hadim, 2016-04-08:
All is in the title.

I have tested it with all the plugin and also a customade plugin (FrameProcessor).

To disable a processor for now I need to remove it.

I couldn't find the code associated to this action... sorry for that.

ChrisWeisiger, 2016-04-08:
Thanks for the report. This has been fixed in 99478973195510d596f538a308e958594ea6f009. For reference, the checkbox was working for newly-created pipelines, but existing ones were not properly being recreated. If you'd closed the Snap/Live window and then re-opened it, it would have picked up the new pipeline.

closed 2016-04-08 by ChrisWeisiger

hadim, 2016-04-08:
Yeah sure I forgot to report the Snap/Lice trick I used.

Thanks