micro-manager/micro-manager #254: Fix ITC18 Device Adapter library issue

kbellve, 2016-04-15:
Replace DeviceAdapter/configure.ac

```
# ITC on linux 
AC_MSG_CHECKING(for ITC)
AM_CONDITIONAL([BUILD_ITC],[test -f "/usr/local/include/ITC18.h"])
if test -f "/usr/local/include/ITC18.h" ; then
   AC_MSG_RESULT([found])
else
   AC_MSG_RESULT([not found])
fi
```

With:

```
# ITC on linux 
AC_MSG_CHECKING(for ITC)

AM_CONDITIONAL([BUILD_ITC], [test -f "/usr/local/include/ITC18.h"])
if test -f "/usr/local/include/ITC18.h"; then
   AC_MSG_RESULT([found])
   if test -z "$LIBITC18"; then
      case $host in
         *-*-linux*)
            LIBITC18="-litc18"
            ;;
      esac
   fi
else
   AC_MSG_RESULT([not found])
fi

AC_SUBST(LIBITC18)
```

Also  please update ITC18 Makefile.am so that the last line reads:
`libmmgr_dal_MMITC18_la_LDFLAGS = $(MMDEVAPI_LDFLAGS) $(LIBITC18)`

nicost, 2020-04-04:
Can't believe this was never done!  Good use of my weekend to go through all these open tickets!

Just pushed this to the subversion repository, but I do not have a Unix system to test.  @kbellve can you check if this works as intended?  @marktsuchida can you make sure this still builds on Mac?

marktsuchida, 2020-04-05:
@nicost The mac build (which doesn't build ITC18) is unaffected by this commit.

kbellve, 2020-04-06:
Thanks for fixing it. I do have an updated version of the ITC18 driver which adds State Device Support. I think probably less than 3 people use the ITC18, and I am one of them, so I never pushed those updates.  On my todo list.

closed 2020-08-28 by marktsuchida