micro-manager/micro-manager #256: UI is unworkable on Windows 10 with high dpi screen

nicost, 2016-04-26:
Both 2.0 and 1.4 have the same problem: the UI is unworkable under Windows 10 with a high resolution screen.  This problem can be fixed by using the launcher executable from the latest ImageJ.  Can the launcher that is currently included be replaced by the one from ImageJ?

Screenshot with current launcher:
[image]

Screenshot of UI with ImageJ launcher:
[image]

ChrisWeisiger, 2016-04-26:
Just to clarify: you mean you went to imagej.net, downloaded the latest version, copied the ImageJ.exe (or more likely, ImageJ-win64.exe) into a µManager installation, and ran that, and it fixes the high-DPI display issue? I don't have a high-DPI display handy to test, but I can confirm that µManager otherwise appears to run normally when I do that, so if that's the appropriate fix then I'm happy to implement it.

nicost, 2016-04-26:
On 4/26/16 8:57 AM, ChrisWeisiger wrote:
>
> Just to clarify: you mean you went to imagej.net, downloaded the 
> latest version, copied the ImageJ.exe (or more likely, 
> ImageJ-win64.exe) into a µManager installation, and ran that, and it 
> fixes the high-DPI display issue?
>

Yes.  I believe that there is only one executable on Windows (i.e. no 
separate version for 32 and 64 bit).
>
> I don't have a high-DPI display handy to test, but I can confirm that 
> µManager otherwise appears to run normally when I do that, so if 
> that's the appropriate fix then I'm happy to implement it.
>

It may be worth checking in the ImageJ github history what change was 
made to the installer, to at least understand what happened. Also, at 
one point we compiled our own ImageJ.exe to include some parameters that 
we needed at that point in time.  I do not know whether or not those 
parameters are still critical under certain conditions.

Best,

Nico

ChrisWeisiger, 2016-04-26:
Thanks for that context. I took a look at the history of the ImageJ.exe file, and it doesn't appear to have changed much since 2010 or so. Unfortunately it is difficult to compare the actual binary vs. other binaries. 

I spoke with Mark Hiner on the ImageJ gitter and he referred me to 

https://github.com/imagej/imagej-launcher/tree/f7d5303a065b171580363f302d3f434a3bd31226/src/main/c

which is the directory that contains the launcher code. There's nothing obvious in the past couple of years that could be related to high-DPI displays, so the bug we're dealing with here may have accidentally been fixed by some other change in the launcher.

ChrisWeisiger, 2016-04-26:
I've updated the launcher in the mm2 branch, so the next nightly build should have the fix. Can you please confirm it when you get the chance? Assuming everything's fine, I'll update the 1.4 branch at that time.

marktsuchida, 2016-04-27:
Can you verify that all the JVM parameters and flags are set correctly when using the new launcher? Also, does the new launcher use the expected JVM?
(I expected we would update our launcher when we switch to Java 8.)
BTW, let's leave the 1.4 launcher alone until we get familiar with the new launcher with 2.0.

nicost, 2016-04-28:
Latest 2.0 version look OK on hdpi Windows.  It uses the JVM that comes 
with MM.  It has the following JPVM flags:

2016-04-27T18:27:34.694449 tid10956 [IFO,App] Java version: 1.6.0_31
                                     [       ] Java runtime: Java(TM) SE 
Runtime Environment 1.6.0_31-b05
                                     [       ] JVM: Java HotSpot(TM) 
64-Bit Server VM; mixed mode
                                     [       ] JVM architecture: 64-bit
2016-04-27T18:27:34.694449 tid10956 [IFO,App] JVM arguments:
                                     [       ]   -Xmx2000m
                                     [       ] -XX:MaxDirectMemorySize=1000g
                                     [       ] -Dpython.cachedir.skip=true
                                     [       ] 
-Dplugins.dir=C:\PROGRA~1\MICRO-~1.0
                                     [       ]   -Xincgc
                                     [       ]   -XX:PermSize=128m
                                     [       ] (End JVM args)

ChrisWeisiger, 2016-04-28:
For comparison, here's the JVM flags from 1.4.23 on our local Windows machine (Windows 7 over remote desktop, no high-DPI display):

Java version: 1.6.0_31             
Java runtime: Java(TM) SE Runtime Environment 1.6.0_31-b05
JVM: Java HotSpot(TM) 64-Bit Server VM; mixed mode
JVM architecture: 64-bit           
JVM arguments:
  -XX:SoftRefLRUPolicyMSPerMB=1000000000000
  -Xmx2000m                        
  -XX:MaxDirectMemorySize=1000g    
(End JVM args)

nicost, 2016-04-28:
I remember that the SoftRefLRUPolicyMSPerMB came from an attempt by Arthur to use soft references for one purpose or another.  I am pretty sure that we are no longer using those, so not having that flag should be fine now. Otherwise, things look the same to me.

marktsuchida, 2016-05-09:
Removing `-XX:SoftRefLRUPolicyMSPerMB` should not be a problem.

ChrisWeisiger, 2016-05-09:
That seems unlikely to be the cause of the DPI problem. Can we keep this open until the new launcher has had a bit more "screen time"? In particular, it'd be nice to eventually backport the new launcher (assuming it is valid) to 1.4.

nicost, 2016-05-09:
New launcher works well for me.  Completely essential on my machine (and likely anyone else with hdpi Windows).  Quite important to backport to 1.4.

marktsuchida, 2016-05-10:
Okay, let's backport to 1.4. Reopening.

ChrisWeisiger, 2016-05-12:
Issue #273 noted that my previous launcher update was incorrect as I'd used the 64-bit launcher, rather than the 32-bit one, thus causing 32-bit MM2.0 to not function. I've replaced the launcher again with the latest 32-bit launcher from imagej.net; could you please check if that has re-introduced the issue on high-DPI displays? If so, we may want to alter our build system to use a separate launcher for 32-bit vs. 64-bit builds.

nicost, 2016-05-16:
Current launcher does not work on Windows 7.  It shows a window saying: "Could not find javaw.exe in the path" or something similar.  Replacing ImageJ.exe in the Micro-Manager directory with ImageJ.exe from the ImageJ download fixes this problem.

ChrisWeisiger, 2016-05-16:
I just downloaded the latest MM2.0 nightly build (64-bit) and was able to launch µManager without any apparent issues. Are you trying to run µManager from the installer, or from your build environment?

nicost, 2016-05-16:
On 5/16/16 1:45 PM, ChrisWeisiger wrote:
>
> I just downloaded the latest MM2.0 nightly build (64-bit) and was able 
> to launch µManager without any apparent issues. Are you trying to run 
> µManager from the installer, or from your build environment?
>

 From the installed version.  This is on Windows 7.  I do install in a 
pre-existing directory (otherwise, I would need to re-install the Andor 
SDK, which is quite involved).

ChrisWeisiger, 2016-05-16:
Could you install into a separate directory just to see if it launches properly? No need to get the Andor SDK installed since the camera isn't needed for this test.

Though, if there's an issue with install-in-place, we still have a problem...just a different one.

nicost, 2016-05-16:
Installed into separate directory: Same problem ("Could not find 
javaw.exe in PATH")

ChrisWeisiger, 2016-05-17:
Okay, thanks for doing all these tests. Very strange; I'm not sure what the difference is between your setup and the one we have here. The most obvious guess would be something in the PATH is different, but I can't imagine where such a difference would come from.

marktsuchida, 2016-05-17:
(I'm going to backtrack on what I said for 1.4. If this can cause unexpected problems on untested systems like this, we should probably keep the 1.4 launcher untouched for the time being.)

nicost, 2016-05-17:
Kurt just asked me for advice because he ran into the same issue.  What 
is different between the ImageJ.exe in the MM and ImageJ distrbution?

ChrisWeisiger, 2016-05-17:
I just went to imagej.net and downloaded the latest 1.x distro, then copied its ImageJx32.exe (or whatever the 32-bit executable is named) into our bindidst/any-Windows directory. Unfortunately it looks like imagej.net is currently down so I can't point you to the specific launcher I used.

The previous launcher that was working for you was the 64-bit launcher, but unfortunately, using that prevents people running 32-bit operating systems from running µManager. One possible solution would be to use a 64-bit launcher in the 64-bit build and a 32-bit launcher in the 32-bit build, but it's still possible that people with 32-bit operating systems would encounter the javaw.exe error.

I don't know what our original launcher (from before this ticket was filed) was; I tried to trace the depository history, but it goes back to the 1.3 days at least, so suffice to say it was pretty old.

nicost, 2016-05-17:
That may be the difference.  I got the ImageJ.exe from the download from 
http://imagej.nih.gov/ij/download.html. Not sure if that one works on 
32-bit Windows though.  I think that it is different from the launcher 
from ImageJ.net.

ChrisWeisiger, 2016-05-17:
After discussion with Mark, I've reverted the launcher changes made for this ticket, so we're back to the launcher we'd been using for the last several years. People using high-DPI displays can download the launcher from ImageJ.net and drop it into their µManager installs for the time being; in the long term we'll need to figure out how to fix this properly, which may involve investigating the details of what exactly these launchers are doing.

marktsuchida, 2016-05-18:
Thanks, @ChrisWeisiger.
Let's revisit this when imagej.net is back up.

@nicost The launcher from Wayne's page appears to predate the recent Java 8 work at imagej.net, and its size is close to the original launcher. Perhaps it is similar to the original launcher and might indeed work (though it's then mysterious how it fixes the high res monitor issue). We'll just need to do some real tests.

Things to test:
- Should work on Windows XP 32-bit, Windows 7 32-bit, Windows 7 64-bit, Windows 10 32-bit, Windows 10 64-bit (I'm willing to skip XP since VMs are no longer available from http://modern.ie (!))
- On Windows 7 and 10 64-bit, both 32-bit and 64-bit versions should work
- Should work on Windows 10 with high res monitor
- Must preserve all (relevant) JVM flags from ImageJ.cfg
- Must launch javaw.exe with working directory set to Micro-Manager directory
- Device adapters with external DLL dependencies in the Micro-Manager directory should load correctly
- JVM system properties such as java.library.path (although should not be critical) should be compared and checked
- Other?

nicost, 2016-05-18:
It is possible that Wayne's launcher uses either or both of the changes described here: http://superuser.com/questions/988379/how-do-i-run-java-apps-upscaled-on-a-high-dpi-display.  Maybe these can be added to the existing launcher?

One other use/improvement for the launcher: display a nice MM icon.

marktsuchida, 2016-05-27:
@nicost That could work (with Java 6 at least). Could you try adding `-Dsun.java2d.dpiaware=false` to your `ImageJ.cfg` (on the 3rd line, right after the `-Xmx` option)?

nicost, 2016-06-02:
That works and fixes the problem (at least for me).  Can this be 
included in the distributions?

ChrisWeisiger, 2016-06-02:
This has been done. Can this issue be closed now?

nicost, 2016-06-22:
I think that this issue can indeed be closed.

ChrisWeisiger, 2016-06-22:
Okay, thanks for confirmation.

closed 2016-06-22 by ChrisWeisiger