micro-manager/micro-manager #257: UNRESOLVED DEPENDENCIES: #iconloader;GIT: not found

pescobar, 2016-04-28:
Hi,

while trying to build micro-manager-2.0.0-beta2 I get an error during the `make fetchdeps` steps. The error message says `UNRESOLVED DEPENDENCIES   #iconloader;GIT: not found`

Here you can see the full output when trying to fetch the dependencies
https://gist.github.com/pescobar/7078e37a92e0dd655d56d31a54b9e0d3

is there any solution for this?

thanks

ChrisWeisiger, 2016-04-28:
You need to update your copy of the 3rdpartypublic repository to include the iconloader jar file. Just go into 3rdpartypublic and run `svn update`, then run `make fetchdeps` again.

ChrisWeisiger, 2016-04-29:
I assume this has been resolved. Please re-open this issue if you continue to have problems.

closed 2016-04-29 by ChrisWeisiger

pescobar, 2016-04-30:
yes this is fixed after copying the 3rdpartypublic folder. Thanks for your help.