micro-manager/micro-manager #258: Error: Could not find or load main class org.micromanager.MMStudio

pescobar, 2016-05-02:
After compiling `micro-manager-2.0.0-beta2` I am getting this error when trying to run the binary:

```
Error: Could not find or load main class org.micromanager.MMStudio
```

Is there any fix for this? Or maybe I did some mistake during compilation?

thanks

ChrisWeisiger, 2016-05-02:
2.0 has a substantially different structure in its codebase. The "central" class is now `org.micromanager.internal.MMStudio` if you need to instantiate it, but otherwise all classes in the `internal` packages are not meant for API consumption (and you should instead work with `org.micromanager.Studio`). See our Javadocs for more information:

http://valelab.ucsf.edu/~MM/doc-2.0.0-beta/mmstudio/

closed 2016-05-02 by ChrisWeisiger

pescobar, 2016-05-02:
HI @ChrisWeisiger 

I am not modifying the code. I just compiled the latest tarball and I get this error when trying to launch the application.

ChrisWeisiger, 2016-05-02:
Ah, my apologies; I assumed you were trying to use a script or run µManager from MatLab. Still, `org.micromanager.MMStudio` definitely does not exist, and there are no valid references to it left in the 2.0 source code (the Magellan plugin refers to it, but Magellan explicitly does not work with 2.0). I expect that instead you have a "stale" launcher left over from 1.4 that still refers to the outdated path, or that your build environment is not correctly set up. Do you have reason to believe that your problem is in fact due to a fault within the source code you downloaded?