micro-manager/micro-manager #260: mm2: add separate pipeline for Live mode vs. MDA

ChrisWeisiger, 2016-05-02:
Users may sometimes want to use a separate processing pipeline for Snap/Live vs. for MDA. For example, #259 (FrameProcessor plugin) does moderately heavy-duty math on images, not to mention changes the semantics of what the time axis means. Rather than having every Processor implement similar "only use in a specific mode" checkboxes, it would be nice if there was an easy way to create separate pipelines for these two use cases. Replacing the current enabled/disabled checkbox in the pipeline GUI with a Live enable/disable and MDA enable/disable might be sufficient.

As an added bonus, that would mean that the MDA and Snap/Live controllers would be responsible for determining whether a given image is part of an MDA or Snap/Live event. This is much cleaner than asking the Processor(s) to do that instead.

hadim, 2016-05-02:
Indeed that would be nice.

ChrisWeisiger, 2016-05-02:
Note that the proposed two-checkbox approach would mean we'd need to resolve what `DataManager.copyApplicationPipeline()` does. Currently it creates a `Pipeline` using only the currently-enabled `Processors`. We could have a separate method each for the Snap/Live pipeline and the MDA pipeline, of course (`copyLivePipeline()` and `copyAcquisitionPipeline()`, say). One potential worry is that if we end up with another dataflow (in addition to Snap/Live and MDA), then this entire GUI and associated API would need to be updated -- and a three-checkbox GUI would be pretty clunky. I'm not convinced that's a very realistic concern, though.

ChrisWeisiger, 2016-07-22:
This has now been done. `copyApplicationPipeline` still exists, and there is now a `copyLivePipeline` as well.

closed 2016-07-22 by ChrisWeisiger