micro-manager/micro-manager #262: Autostretch feature takes into account ROI display on the image window

hadim, 2016-05-04:
I have the feeling the Autostretch feature takes into account the pixel drawn when using the ROI tools for example and so the contrast scale become totally wrong when using such a tool. It's harder to reproduce with the demo config but I totally see it with all the microscopes I use.

Step to reproduce : 

- launch a live window
- check the Autostretch checkbox 
- select the rectangle tool from the ImageJ window
- draw a rectangle on the live window
- the autostretch should start scaling incorrectly the image

(I noticed that using mm2, I didn't try but I am pretty sure I used to saw that also with mm1).

ChrisWeisiger, 2016-05-04:
This is working as intended. If you have an ROI on the image, then only the pixels in the ROI are a) shown in the histogram, and b) used to calculate autostretch. Or am I misunderstanding what you're saying?

hadim, 2016-05-05:
Sorry I wasn't clear I think. I am saying that the pixel **of the ROI** (the yellow pixel) used to draw the ROI are taken into account and so the scale become very weird when adding a rectangle or whatever ROI it is on the image window (I am not talking here about the use of a ROI for acquisition on the camera only ROI drawn on the image window).

hadim, 2016-05-05:
Actually you can use the demo config to show something similar as I have on the microscope by pressing the left button of the mouse when selecting an area with a ROI (whatever the shape is). When the left button is pressed you can notice the min and max of the histogram being set to -1.

hadim, 2016-05-05:
Oh no sorry I get it now... the area covered by the ROI drawn is the one used for the calculation of the min/max... I didn't know that.... That is why sometime when selecting a very specific region of my sample I had weird histogram not scaling with my whole sample.

hadim, 2016-05-05:
The thing is sometime I draw a ROI only to put a marker over a spot or a feature of my sample (to see how it moves actually) and I don't want necessarily to use the pixels of the ROI to compute the histogram. 

Would you integrate it if I add a checkbox on the histogram panel to enable/disable this feature ? I you do, could you point me out to the right class to modify, it will save me sometime.

ChrisWeisiger, 2016-05-05:
Okay, I'm glad we're clear on what's going on. :)

A toggle option to enable/disable this sounds reasonable. I would suggest adding it to the "gear menu" for the histograms panel, which is where we've put a number of other rarely-accessed configurable settings. Said menu is generated by `getGearMenu` in

https://github.com/micro-manager/micro-manager/blob/mm2/mmstudio/src/main/java/org/micromanager/display/internal/inspector/HistogramsPanel.java

Suggested option text: "Use ROI When Scaling".

You would also need to add a DisplaySettings property to reflect this setting; something like a Boolean `shouldScaleWithROI` sounds appropriate. Make certain you handle saving/loading this property in `DefaultDisplaySettings.get/setStandardSettings` and in `DefaultDisplaySettings.legacyFromJSON/toJSON`. 

Finally, ContrastCalculator will need to be modified to take the property into account:

https://github.com/micro-manager/micro-manager/blob/mm2/mmstudio/src/main/java/org/micromanager/display/internal/ContrastCalculator.java

Be aware that ContrastCalculator is somewhat tightly optimized because histogram calculations are expensive. You should restrict yourself to modifying the code in the InternalCalculator constructor that determines whether there's an ROI or not. Do not modify the various calculate* methods.

Let me know if you have any questions.

hadim, 2016-05-05:
See the PR.

closed 2016-05-05 by hadim