micro-manager/micro-manager #264: Acquisition Engine: Z positions are inaccurate during sequenced Z stacks

ChrisWeisiger, 2016-05-05:
Repro case: load demo config, run a Z-stack. Note that the ZPositionUm metadata field for images is accurate. Now set the "UseSequences" property of the Z stage to Yes, and run the same Z-stack. Note that the ZPositionUm metadata field now shows the "top" position of the stack for every image in the stack.

Due to the way sequencing works, it is not possible for us to provide accurate (i.e. measured) Z positions for each image in a sequenced stack. However, we should not show known-inaccurate information, so at minimum the ZPositionUm field should be blanked for all (or all but the first?) images in the stack. We could potentially add an IntendedZPositionUm field to the metadata to indicate the position that we wanted to use. If we did this, it would be nice if that field were used as a fallback in the GUI (with some indicator, like a "?", that the field's accuracy is unknown). 

This issue applies to both 1.4 and 2.0.

marktsuchida, 2020-08-28:
Still reproducible today.

Splitting "setpoint" and "measurement" is a good idea beyond this issue.

nicost, 2023-07-05:
This is handled more nicely in AcqEngJ.