micro-manager/micro-manager #270: Live mode is slow with Andor iXon camera

nicost, 2016-05-11:
Running an Andor iXon in live mode with settings that allow for maximum frame rate (~~30 fps, i.e. 14--bit digitizer, 10 MHz readout rate, overlap mode on) results in frame rates of about 6fps (Data 6fps, Display 12.5 fps).  Live mode in 1.4 is 30fps as expected.

ChrisWeisiger, 2016-05-11:
This is just a 512x512 camera, right? Nothing special there, no reason why the display would be choking us up.

If I use the DemoCamera and set its ReadoutTime property to 30, then I can run live mode at 28FPS (data and display). That ought to simulate the iXon running an internal sequence acquisition.

Best guess is that the thread grabbing images for live mode thinks it's not capable of achieving 30FPS for some reason. We have some throttling logic in there so that e.g. if you have a pipeline doing complicated processing, we won't try to shove images down the pipeline faster than they can be processed. But without a repro case, this is going to be hard to test.

Do you have a processing pipeline set up at all?

nicost, 2016-05-11:
On 5/11/16 11:30 AM, ChrisWeisiger wrote:
> This is just a 512x512 camera, right? Nothing special there, no reason
> why the display would be choking us up.
>
> If I use the DemoCamera and set its ReadoutTime property to 30, then I
> can run live [...]
No, none.

If you want, I can make sure the development environment is setup and 
give you a TeamViewer session.  This problem/bug makes 2.0 unusable for 
quite a few applications.

It is of course also possible that some camera settings are not what 
they appear to be.  Still, very strange that all is good in 1.4.

marktsuchida, 2016-05-11:
Given that we looked at display frame rates a while back (and @ChrisWeisiger optimized histogram code), I wonder if this is a regression. Perhaps bisect 2beta nightlies from the past few months?

ChrisWeisiger, 2016-05-12:
Mm, I don't think that recent changes can explain this. The FPS calculations use as numerator either (number of images inserted into datastore) or (delta in image sequence numbers), preferring the latter when available. Which it should be in an iXon sequence acquisition. Similarly, the denominator is either (number of display updates performed) or (delta in image elapsed time MS), preferring the latter, which should be available in an iXon sequence acquisition.

In other words, if you look at the image metadata in a live acquisition and see both the ImageNumber and ElapsedTimeMs fields being populated, then the data FPS really should be accurate and the iXon is actually only acquiring ~6 images per second.

What happens if you do a sequence MDA?

ChrisWeisiger, 2016-05-12:
I discovered a bug in `AcquisitionManager.generateMetadata()` that caused the `imageNumber` and `elapsedTimeMs` properties to be lost; that bug is fixed in 8d0e4dc64f926f57d455f024388e8c4f5386b510. This should fix issues with the FPS display, though it shouldn't affect the actual mechanics of grabbing and displaying Live images. Does it help any with your issue?

nicost, 2016-05-12:
Bit busy right now, but I'll try to test tomorrow. Will also test older 
builds when I have a chance.

ChrisWeisiger, 2016-05-12:
Sure, no rush. Thanks for being our tester. :)

nicost, 2016-05-16:
On a second system with Andor iXon using the latest build I see that Data fps 34 fps and Display fps 4 fps.  Even when switching Histogram update off there is no speed up.  There is no appreciable cpu usage (1-2%) while live mode is running. Any ideas?

nicost, 2016-05-18:
On the same computer, when using the demo configuration, live mode is as expected (i.e. data at 100 fps, display at 30 fps).  Hence, I do believe that there is an issue specific to the Andor camera (but strangely enough, the issue only shows up in 2.0, not in 1.4).  Nevertheless, it is debilitating enough to make 2.0 unusable (focusing is very difficult at 4 fps).

ChrisWeisiger, 2016-05-18:
I'd be inclined to suspect changes in the C++ layer, but MM2's C++ layer is up to date as of May 12th, so that seems unlikely. It remains possible that there's a bug in the Snap/Live code that I've missed so far; what happens if you run a sequence MDA?

nicost, 2016-05-18:
Just tested with the nightly 1.4 to make sure that I did not miss anything.  1.4 is fine.  MDA in 2.0 is also fine, it acquired 100 images in 3 seconds (which is exactly as expected), and also displayed the images at full speed during acquisition.  It is quite strange that live mode in 2.0 does not keep up with a real camera but does with the demo camera....

ChrisWeisiger, 2016-05-18:
Okay, let's assume there's a bug with the 2.0 Live image management that's causing it to excessively throttle the display. I'll look into it. If I'm not able to figure anything out from just looking at the code, we may need to arrange a remote login session.

ChrisWeisiger, 2016-05-18:
I fixed a bug (more like a misbehavior) in `AcquisitionManager.generateMetadata()` that caused it to pull information from the hardware for every image acquired during live mode. It's conceivable that this slowed things down on your system enough to seriously impact the display rate. Could you please try tonight's nightly build?

If that still shows problems, could you please add a log line after line 256 of `org.micromanager.internal.SnapLiveManager` to log the values of `average` and `displayRate`? That's where we calculate how long to sleep in between each image display.

nicost, 2016-05-19:
Build MM from source on one of the microscope computers.  Issue is not fixed.  studio_.logs().logMessage(String) does not seem to log messages to either the Corelog or the Netbeans output.  Old fashioned System.out.println() shows:
Average: 0, displayRate: 33
Average: 0, displayRate: 33
Average: 33, displayRate: 33
Average: 71, displayRate: 53
Average: 78, displayRate: 58
Average: 85, displayRate: 63
Average: 89, displayRate: 66
Average: 93, displayRate: 69
Average: 95, displayRate: 71
Average: 97, displayRate: 72
Average: 98, displayRate: 73
Average: 100, displayRate: 75
Average: 102, displayRate: 76
Average: 104, displayRate: 78
Average: 105, displayRate: 78
Average: 106, displayRate: 79
Average: 107, displayRate: 80
Average: 108, displayRate: 81
Average: 108, displayRate: 81
Average: 109, displayRate: 81
Average: 110, displayRate: 82

nicost, 2016-05-19:
Also, commenting out the sleep just below that line does not make a difference whatsoever.

ChrisWeisiger, 2016-05-19:
That is incredibly bizarre. Those values indicate that the display _is_ being updated a bit under 10x/second (that's what the "average" is indicating). These values are calculated based on when the display publishes `PixelsSetEvent`s, which should be a very reliable way of saying "what is on screen has been changed". And yet you're only seeing 4 "real" updates per second, and the display FPS in the display window also indicates 4? Please verify my understanding is accurate.

I could conceive of a bug causing us to repeatedly grab the same image from the sequence buffer, thus creating an apparent slow display update rate (as the displayed image wouldn't change), but the FPS calculation would still show a high value in that case (we'd be repeatedly drawing the same image). I can't figure out how to reconcile a calculated update rate in the `SnapLiveManager` of ~10FPS with a calculated update rate in `HyperstackControls` of ~4FPS. The two do use slightly different events (`PixelsSetEvent` vs. `CanvasDrawCompleteEvent`), but said events should have a 1:1 correspondence. 

I also can't think of any way that the hardware could be at fault without also impacting MDA. You've found a very strange bug here.

nicost, 2016-05-19:
> That is incredibly bizarre. Those values indicate that the display
> /is/ being updated a bit under 10x/second (that's what the "average"
> is indicating). These values are calculated based on when [...]
Sounds about right, although it is possible that the displayed fps on 
this particular run was somewhat higher.  Nevertheless, a bit under 
10x/second, is still a lot less than 30x/second, which it easily does on 
1.4.  I also noted that on longer runs the average number seems to go 
up, so I would not focus too much on that difference, rather that both 
10fps and 4 fps are significantly slower than 30 fps (which has to be 
the goal here as 1.4 easily manages that rate, and the hardware easily 
gets that rate).
> I could conceive of a bug causing us to repeatedly grab the same image
> from the sequence buffer, thus creating an apparent slow display
> update rate (as the displayed image wouldn't change), but [...]
I can not oversee/get in the code leading up to the onDisplayUpdated 
call (that remains one of the issues I dislike with callbacks), but 
something in that chain is clearly too slow.
>
> I also can't think of any way that the hardware could be at fault 
> without also impacting MDA. You've found a very strange bug here.
>
Could there be a thread priority or locking issue here with the camera 
thread taking priority and leaving little time for the display to 
update?  I could imagine that the fast update of the MDA window actually 
happens after the camera finishes its sequence acquisition.

In any case, it seems quite essential that this one gets addressed. Is 
there any way that you can use a real camera on your system (preferably 
on Windows, preferably an Andor)?

ChrisWeisiger, 2016-05-19:
On Thu, May 19, 2016 at 8:50 AM, Nico Stuurman <notifications@github.com>
wrote:

> Sounds about right, although it is possible that the displayed fps on
> this particular run was somewhat higher. Nevertheless, a bit under
> 10x/second, is still a lot less than 30x/second, which it [...]

If we're actually achieving 10x/second, then that at least makes the code
internally consistent, which greatly simplifies the problem scope. There's
still clearly a problem, but it's one of optimization rather than bizarre
inconsistencies.

> I can not oversee/get in the code leading up to the onDisplayUpdated
> call (that remains one of the issues I dislike with callbacks), but
> something in that chain is clearly too slow.
>

That is one side-effect of the event system, yes -- it decouples code and
makes calling chains less obvious. If you don't have a stack trace handy,
you can search for places where the event is created, since in most cases
there should only be one originator of the event. In this case, you would
search for DefaultPixelsSetEvent, the implementation of the PixelsSetEvent
interface.

> Could there be a thread priority or locking issue here with the camera
> thread taking priority and leaving little time for the display to
> update? I could imagine that the fast update of the MDA [...]

This should be testable by running longer MDAs, so that the display is
still active while the camera is acquiring. I suppose that this issue could
be explained by resource (CPU, RAM, etc.) contention.

>
> In any case, it seems quite essential that this one gets addressed. Is
> there any way that you can use a real camera on your system (preferably
> on Windows, preferably an Andor)?
>

I'll talk to the TI folks and see if something can be arranged. Debugging
over email is not very efficient.

ChrisWeisiger, 2016-05-19:
I've fixed one more bug in `AcquisitionManager.generateMetadata()` -- the pixel size was being pulled from the core (which involves acquiring a lock on the camera device so that the binning can be determined), instead of using a cached value. Mark and I have looked over the method and are now confident that it's not going to the core for anything. Could you try tonight's nightly build?

If that doesn't fix it, could you also try a longer MDA (e.g. 1000 frames) to see what the long-term FPS are?

ChrisWeisiger, 2016-05-19:
You could also try turning off autostretch; we've discovered a bug with it that causes excessive redraws. I don't _think_ that it's applicable to this issue, but it's an easy experiment to try.

nicost, 2016-05-20:
Did that already.  Has no influence.

ChrisWeisiger, 2016-05-20:
Okay, please do still try tonight's build when you get a chance, though.

ChrisWeisiger, 2016-05-20:
Okay, I've found yet _another_ place where Live mode repeatedly acquires a lock on the camera device -- when it checks the number of core camera channels, which it currently does once per image. So you can hold off on testing for the moment.

ChrisWeisiger, 2016-05-20:
Fix pushed; as far as I can tell the only core methods that are repetitively called when running live mode now are ones that do not acquire locks on anything (`getNBeforeLastTaggedImage()`, `getCurrentConfig()`, `getChannelGroup()`, `getSystemStateCache()`, `getPropertyFromCache()`). Please test at your convenience.

nicost, 2016-05-20:
Using yesterday night's build, live mode speed is much better.  With autostretch on, I get 30fps, with autostretch off, however, the rate drops to 10 fps.  This seems related to issue #278.  However, with autostretch on, I do not believe there are excessive screen draws as the display is visibly faster then with autostretch off.  This rather suggests that the autostretch off code execution path does something bad.

B.t.w., whether or not a channlegroup is selected does not seem to make a difference.

Not sure, why, but I am unable to run the current source code as Netbeans seems to execute the code in MMJ_ rather than the source code.  It would be great if you happen to know a fix for that problem.

ChrisWeisiger, 2016-05-20:
Thanks for the update. Does the display visibly look different with autostretch on vs. off? Assuming our hypothesis in #278 is accurate, the extra redraws won't actually show you new images (i.e. they're drawing the same image twice). If you are actually getting a smoother-looking display with autostretch on, then something else must be going on.

Thanks for checking the channelgroup, since I'd been meaning to ask you about that. :) That's helpful information.

Re: Netbeans, I seem to recall that we discussed an issue like that previously over email. You might want to double-check the library jars in your project properties, and also make certain you don't have any stale copies of MMJ_.jar lying around. We haven't changed the build system in ages, so I'd suspect something unusual in your build/deploy environment is causing problems.

nicost, 2016-05-20:
Worked around the Netbeans issue (by putting MMJ_.jar in the "Libraries" "Run" tab under "Compiled Sources".  Effect of checking/unchecking Autostretch is very obvious is output of waitTime and shortestWait.  For instance: 

waitTime: 55, shortestWait: 74  (autostretch off)
waitTime: 33, shortestWait: 19 (autostretch on)

Also, when starting live mode with autostrech off, display fps starts at ~2fps and then slowly ramps up to ~10fps. A ramp up is also visible with autostrech on, but it goes much faster to 30fps.  The initial slowness may be explained by the following output:
waitTime: 500, shortestWait: 9223372036854775807
waitTime: 500, shortestWait: 9223372036854775807
waitTime: 33, shortestWait: 30
waitTime: 33, shortestWait: 30

A bit of a similar effect can be seen in MDA (1000 images, 0 ms interval, 0 ms exposure).  Autostretch on leads to 45 fps (not sure how that is possible, isn't display limited at 30fps?)  Autostretch off ~32fps.

ChrisWeisiger, 2016-05-20:
Note that the display FPS is a measurement of how many times the image canvas draws per second, so e.g. you can get high display FPS by simply turning overlays on and off or by scrolling through a dataset. Or by a bug in the software simply trying to draw the same image over and over again, which is what we assume is happening here. Live mode normally tries to rate-limit itself to 30FPS, and of course your data source is only supplying 30FPS, but there's nothing that says we can't try to draw those images more frequently. You can see high display FPS values by simply turning an overlay on and off, or by scrolling through a dataset -- all the display FPS is measuring is the number of times the image canvas draws per second.

Thanks for pointing out the bug with shortestWait; I have pushed a fix.

nicost, 2016-05-20:
> Thanks for the update. Does the display visibly look different with
> autostretch on vs. off? Assuming our hypothesis in #278
> <https://github.com/micro-manager/micro-manager/issues/278> is [...]

Yes, the display looks like it shows more images per second when 
autostretch is on versus off, i.e. I do not think that the same image is 
drawn twice when autostretch is on

ChrisWeisiger, 2016-05-24:
I've fixed the autostretch-redraw bug, which will unfortunately presumably worsen your display update rate. I don't want to "fix" one bug by allowing another bug to remain in the codebase, though.

I still need to get in contact with TI to see if they have an iXon we can borrow. In the meantime, if you want to collect some more diagnostics, logging when exactly the `CanvasUpdateQueue.enqueue()` and `CanvasUpdateQueue.onCanvasDrawComplete()` methods are called may shed some light on how long drawing is actually taking us.

nicost, 2016-12-20:
Closing since the display code will be re-written and will need re-testing.

closed 2016-12-20 by nicost