micro-manager/micro-manager #271: Can user profile in splash screen default to the last one used?

nicost, 2016-05-11:
When restarting 2.0 (which I have to do quite regularly), I now every time have to select my own user profile, which I of course forget quite often.  Can the user profile default to the last one used rather than the default profile?

ChrisWeisiger, 2016-05-11:
Why do you have to restart 2.0? If there's an issue requiring frequent restarts, it'd be good to know about.

Making the default profile not be "Default profile" but rather the last-used profile is something I have mulled over in the past. I haven't come to a conclusion one way or another yet, but I'll definitely keep your views in mind.

nicost, 2016-05-11:
On 5/11/16 11:51 AM, ChrisWeisiger wrote:
>
> Why do you have to restart 2.0? If there's an issue requiring frequent 
> restarts, it'd be good to know about.
>
Nothing specific to 2.0.  While testing/setting up things, frequent 
restarts are the norm.

> Making the default profile not be "Default profile" but rather the 
> last-used profile is something I have mulled over in the past. I 
> haven't come to a conclusion one way or another yet, but I'll 
> definitely keep your views in mind.
>
I also agree that this should not be something automatic, i.e. users 
should remember to check.  Problem is that everyone now clicks through 
that window as if it did not exist (I am guilty myself there).  
Nevertheless, it would be nice that the same profile is pre-selected 
when restarting MM.

Would it be nutty to pre-select the previous profile if MM was quite 
within ~ 1 hour and otherwise select the default profile?

marktsuchida, 2016-05-11:
One vote against a mysterious timed switch ;)

One thought: we could have an option to present the popup with no selection initially, and requiring the selection before the OK button can be clicked. This would perhaps be a checkbox "Pre-select Default user profile" in Options, checked by default. This would be advantageous even for non-developers in some situations.

I'm against adding any "memory" for the last-used profile, since only developers coding on a shared computer account should be wanting that. We could, as a developer-only feature, have a hidden setting (e.g. a Java system property `-Dmm.userprofile.default=Nico`) to auto-select a non-Default profile, though.

closed 2020-04-20 by nicost