micro-manager/micro-manager #272: User profile display in main windows and in metadata

nicost, 2016-05-11:
After the splash screen there is no feedback about the user profile that was used.  Can that be added somewhere in the main window?  Also, at the AQLM course someone made a point of adding the name of the operator to the image metadata.  Is the name of the profile added to the metadata, and if not, can this be done easily?

ChrisWeisiger, 2016-05-11:
There's a `profileName` property in the `SummaryMetadata` which contains the name of the user profile being used. I think it'd be redundant to add it to each image's metadata.

Good call on indicating what user is being used somewhere in the GUI. Maybe in the title of the main window?

nicost, 2016-05-11:
On 5/11/16 11:32 AM, ChrisWeisiger wrote:
>
> There's a |profileName| property in the |SummaryMetadata| which 
> contains the name of the user profile being used. I think it'd be 
> redundant to add it to each image's metadata.
>

Perfect!  I simply had not yet looked for it.

> Good call on indicating what user is being used somewhere in the GUI. 
> Maybe in the title of the main window?
>

Either there, or in whatever empty space is available in the main window.

ChrisWeisiger, 2016-05-11:
I've added it to the title of the main window, after the config file, in 34d3024b9b078d70a13be5ec427f512e095df6b0. That title is now pretty extensive, though, so it's possible the name could run off the end of the window if the config file path is lengthy.

I'm not sure where we could put it in the window's contents itself. The only free space that I can see is near where the cite plea goes, and we have plans for that space already.

marktsuchida, 2016-05-11:
Agreed that the title is too crowded (even before adding the profile, sometimes).
Could we maybe add a line at the top of the window for the profile (left-justified) and config file (right-justified, with middle truncated if necessary)?
Another possibility is the bottom of the window (second line), but that might make it harder to read the (more scientifically relevant) existing status line.

ChrisWeisiger, 2016-05-12:
Here's what Mark's suggestion looks like:

[image]

It feels a little weird to me to not have the Snap and Save buttons be the topmost things in the window, but I'm sure I'd get used to it with time. Opinions from you two? If we add this, should we remove the config file and profile name from the title?

nicost, 2016-05-12:
I like it, and removing the config file and profile from the title will 
make the window a lot cleaner. Does it look good in the Day scheme?

ChrisWeisiger, 2016-05-12:
See for yourself:

[image]

This also shows the stripped-down title, and having seen that, I think that the improvement in the title is definitely worth moving the config file and profile into the window proper.

ChrisWeisiger, 2016-05-12:
Went ahead and pushed the updated layout as per the above commit. I think we're done here.

closed 2016-05-12 by ChrisWeisiger

marktsuchida, 2016-05-12:
In terms of visual appearance, a thin, gray line separating the line from the rest of the window might make it look better. (Actually might also be good for the bottom status line.)

ChrisWeisiger, 2016-05-12:
Done, and switched the positions so the config file is above the config presets.