micro-manager/micro-manager #273: mm2: 32bit MM build are broken.

hadim, 2016-05-12:
When I try to launch MM I have this message : "ImageJ.exe is not a valid Win32 application". I guess something went wrong when you recently replace the launcher.

ChrisWeisiger, 2016-05-12:
Can confirm, though the error message is different on our 64-bit system; it launches fine but isn't able to load the 32-bit libraries. My mistake for updating to the 64-bit launcher. Should be fixed now, though it'd be good to now check up on issue #256 which the launcher update was originally intended to fix.

Please confirm if this issue can be closed.

hadim, 2016-05-12:
I confirm the fix. (the IJ icon does not show anymore on the .exe, not a big deal...).

closed 2016-05-12 by hadim