micro-manager/micro-manager #274: Inspector window is always on top

nicost, 2016-05-12:
Maybe this is heretic, but now working with 2.0 more regularly, I start to dislike the always on top behavior on the Inspector window.  It is difficult to arrange windows around it, so that their content becomes visible.  I do appreciate our consideration that it is an important window, and that it should stand out, but currently, it is more bothersome then helpful.  Not super important, but thought it useful to bring this up.

ChrisWeisiger, 2016-05-12:
One of the common complaints about 1.4 was that when you get more than a few image windows open, it becomes impossible to keep easy access to the main window where your histograms and metadata are. Making the inspector be always-on-top is our solution to that problem. Can you propose an alternative solution?

hadim, 2016-05-12:
Maybe a big visible button on the main MM window that will bring the Inspector window on top once clicked ?

nicost, 2016-05-12:
Hey, I am not the UI expert, just a user;)  I can think of various approaches.  You could bring the Inspector to the front whenever a new image window is opened.  There could be a button on the image window to bring the inspector to the front.  The problem with the Inspector always-on-top is that the Inspector window can get quite large and leave little space on the screen for plugins, images, etc.. Also, its behavior to disappear whenever Micro-Manager goes out of focus (i.e. when switching to another application) makes it harder to work with it.

ChrisWeisiger, 2016-05-12:
@hadim Keep in mind that the main window also does not stay on top, so you're still stuck with trying to find a specific window (or with your suggestion, one of two specific windows) in your massive pile.

We might be able to rig up something on the display window to let you raise _an_ inspector, but keep in mind there is no "_the_" inspector window. We create one automatically when you first create an image window, but the user is free to close that, create their own, etc.

What kind of display are you working with? I'll freely agree that on my 13" Macbook, the inspector frequently gets in the way, but I'm not trying to drive a microscope on this system. I'd expect most of our users will be working with a 21" display at minimum.

I'd prefer if the inspector stay visible when switching applications, but unfortunately the "disappear when focus switches" behavior seems to be something at the operating system level, so we're unlikely to be able to fix that.

nicost, 2016-05-12:
I am using it on 21" displays on microscopes, and find it to be as hard to work with as on smaller displays.  I completely agree that the idea is good, but in practice it does not work so well.

hadim, 2016-05-12:
I know but (at least for me) I always try to keep the main window at the same place without any other window on top of this one. So it's always visible.

marktsuchida, 2016-05-12:
I think that the current behavior is just a little short of being the correct behavior for a floating window. Well-designed apps that show floating windows usually try not to pop up normal windows under the floating window. Might it be possible to change the auto-positioning of new image windows so that they don't get obscured by the Inspector?

Ideally it would also be easier to dock the Inspector to the right (or left) edge of the monitor.

I also wouldn't be totally opposed to adding an option that allows the user to switch between floating and normal window for the Inspector (useful on a laptop). When I proposed the floating Inspector, we were not expecting to have more than one Inspector open at a time. But I think better positioning of new image windows should be considered even then (at least when the inspector is close to the edge of the screen).

The main idea behind the Inspector's tall shape was that it could take advantage of the aspect ratio mismatch between modern LCDs (wide) and many scientific camera images (closer to square), so that the Inspector typically doesn't overlap the image windows. If that's not happening naturally, there's probably room for improvement. (This applies whether or not the Inspector is floating.) Unless we're talking solely about cases where the user is viewing two images side by side (are we?).

closed 2016-08-03 by marktsuchida