micro-manager/micro-manager #276: ImageJ focus is always with the last image opened

nicost, 2016-05-13:
Open two datasets.  Switch focus to the first data set opened.  Use an ImageJ command (such as "Duplicate").  Notice that ImageJ executes this command on the second data set that was opened, not the one that has focus.  This is also the case with datasets that result from starting an acquisition.  

A possibly related bug (which can be quite disastrous), happens when running multiple acquisitions in sequence.  Then, try to close older acquisition windows while running an acquisition.  Micro-Manager apparently thinks that the user tries to close the current acquisition window and offers to stop the acquisition.

hadim, 2016-05-14:
I have also noticed the same kind of bug on mm2 few weeks ago without taking the time to report it.

ChrisWeisiger, 2016-05-16:
Both issues have now been fixed. Thanks for the reports!

closed 2016-05-16 by ChrisWeisiger