micro-manager/micro-manager #279: which java version should I use to compile on linux?

pescobar, 2016-05-30:
Hi,

I have compiled micromanager in centos6 as a ImageJ plugin using java7 but I have noticed that ImageJ provides a JRE which is 1.6.0_24 

```
[micro-manager-test-install]$ ./ImageJ/jre/bin/java -version
java version "1.6.0_24"
Java(TM) SE Runtime Environment (build 1.6.0_24-b07)
Java HotSpot(TM) 64-Bit Server VM (build 19.1-b02, mixed mode)
```

which is the recommended java version I should use to compile micro-manager on linux? is 1.7 fine or should I use 1.6?

ChrisWeisiger, 2016-05-30:
µManager has only been extensively tested on Java 6. We are hoping to move to Java 8 at some point, but haven't made firm plans. If you're willing to be a tester, we'd appreciate knowing what if any issues you encounter using a more recent Java.

pescobar, 2016-05-31:
hi @ChrisWeisiger 

thanks for the clarification. I am not a micro-manager user so I cannot test. I have to install micro-manager for my cluster users so I should provide it with the best supported dependencies. 

thanks

closed 2016-05-31 by pescobar