micro-manager/micro-manager #283: Cannot find FrameProcessor plugin

highwind888, 2016-06-03:
Newbie here, so not sure what I'm doing wrong. I installed Micro-Manager 1.4, and need to use the FrameProcessor plugin. It was not in the Plugins menu despite being standard; when I downloaded the .jar file and copied it to the mmplugins folder, it still does not appear. I have tried putting it into a subfolder, and while the subfolder is present, the plugin is not. When I try to move it while MicroManager is running, I am prevented from doing so because it is "currently open".

It seems to be loaded, but do I need to activate it somehow before I can get it to appear?

hadim, 2016-06-03:
`FrameProcessor` is not available for MM 1.4 but only for MM 2 and above. If you don't want or can't switch to MM 2 you can `FrameAverager` as an alternative : http://www.openpolscope.org/pages/MMPlugin_Frame_Averager.htm

ChrisWeisiger, 2016-06-03:
hadim is correct. FrameProcessor is built into µManager 2.0, but is not available in µManager 1.4. Note that as of tonight's nightly build (dated 20160603) it will instead be named "FrameCombiner". The functionality however is unchanged.

highwind888, 2016-06-06:
Thank you very much for the prompt response. I should have checked the compatibility first! I will do as you advise

ChrisWeisiger, 2016-06-06:
I'm afraid we can't help you with difficulties with the Frame Averager plugin. I recommend contacting the people who wrote it. You should also provide them with the CoreLog corresponding to your attempt to use the plugin (you can find your logs in the "CoreLogs" directory of your µManager installation), as it may have some information they can use to figure out what's going wrong.

Good luck!

closed 2016-06-09 by ChrisWeisiger