micro-manager/micro-manager #287: Warning at MM launch with Windows 10 & Netbeans

hadim, 2016-06-09:
All is in the title. See the log below.

```
Jun 09, 2016 10:13:55 AM java.util.prefs.WindowsPreferences WindowsRegOpenKey1
WARNING: Trying to recreate Windows registry node Software\JavaSoft\Prefs at root 0x80000002.
Jun 09, 2016 10:13:55 AM java.util.prefs.WindowsPreferences openKey
WARNING: Could not open windows registry node Software\JavaSoft\Prefs at root 0x80000002. Windows RegOpenKey(...) returned error code 2.
2016-06-09T10:13:55.609572 tid1476 [IFO,App] Error accessing old user preferences
                                   [       ] java.util.prefs.BackingStoreException: Could not open windows registry node Software\JavaSoft\Prefs at root 0x80000002. in Thread[AWT-EventQueue-0,6,main]
                                   [       ]   at java.util.prefs.WindowsPreferences.childrenNamesSpi(WindowsPreferences.java:751)
                                   [       ]   at java.util.prefs.AbstractPreferences.childrenNames(AbstractPreferences.java:718)
                                   [       ]   at java.util.prefs.AbstractPreferences.getChild(AbstractPreferences.java:1259)
                                   [       ]   at java.util.prefs.AbstractPreferences.nodeExists(AbstractPreferences.java:898)
                                   [       ]   at java.util.prefs.AbstractPreferences.nodeExists(AbstractPreferences.java:878)
                                   [       ]   at org.micromanager.internal.utils.DefaultUserProfile.getLegacyPreferences14(DefaultUserProfile.java:898)
                                   [       ]   at org.micromanager.internal.utils.DefaultUserProfile.getLegacySystemPreferences14(DefaultUserProfile.java:892)
                                   [       ]   at org.micromanager.internal.dialogs.RegistrationDlg.getHaveRegistered(RegistrationDlg.java:262)
                                   [       ]   at org.micromanager.internal.menus.HelpMenu.initializeHelpMenu(HelpMenu.java:39)
                                   [       ]   at org.micromanager.internal.MMStudio.<init>(MMStudio.java:318)
                                   [       ]   at MMStudioPlugin$1.run(MMStudioPlugin.java:82)
                                   [       ]   at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
                                   [       ]   at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
                                   [       ]   at java.awt.EventQueue.access$500(EventQueue.java:97)
                                   [       ]   at java.awt.EventQueue$3.run(EventQueue.java:709)
                                   [       ]   at java.awt.EventQueue$3.run(EventQueue.java:703)
                                   [       ]   at java.security.AccessController.doPrivileged(Native Method)
                                   [       ]   at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
                                   [       ]   at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
                                   [       ]   at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
                                   [       ]   at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
                                   [       ]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
                                   [       ]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
                                   [       ]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
                                   [       ]   at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

Note that I also use Java 8 (impossible to install Java 6 on Windows 10 because of error 1723...).

ChrisWeisiger, 2016-06-09:
This shows up on other systems as well, and should probably not be logged as it's not a big deal. It just indicates that we were unable to find old (1.4-style) preferences for your account. 

I've gone ahead and pushed a commit that removes the error spew.

closed 2016-06-09 by ChrisWeisiger

hadim, 2016-06-09:
Nice. Thanks.