micro-manager/micro-manager #296: mm2 repository contains non-functional plugins

nicost, 2016-06-16:
There are a number of plugins in the mm2 github repository that have not been updated to version 2.0.  Examples are "Example" and "Hello World" (which are the first ones I look at when writing a new plugin).  There are likely more.  Can these be removed or updated as appropriate?

ChrisWeisiger, 2016-06-16:
The disabled plugins are DLLAutoReloader, DemoCustomDisplay, Example, HelloWorld, Magellan, and TwoPhoton. DLLAutoReloader is intentionally deprecated, DemoCustomDisplay is already up-to-date, Magellan is Henry's responsibility (considering the way it is designed and the fact that it is under active development, updating it to the 2.0 API is not feasible), and TwoPhoton is disabled in 1.4 as well, as far as I can tell.

I've updated Example and HelloWorld to the 2.0 API. I'm not convinced we need both of them, though, and HelloWorld seems like a simpler example. Example does demonstrate how to create a basic GUI, but that's such a complicated topic that there's no way we can possibly do it justice. Is there a compelling reason to keep it beyond that its name is a bit more indicative of what it does than the HelloWorld plugin's is?

marktsuchida, 2016-06-16:
I think we can delete DLLAutoReloader, Magellan, and TwoPhoton from mm2.

DLLAutoReloader was only ever useful under very narrow conditions that are hard to firmly describe (the structure of the device adapter module being reloaded is part of the conditions). Magellan and TwoPhoton can be revived whenever they are ported to mm2.

I think Example is also no longer needed, at least in its current form that uses NetBeans interface builder (something that we no longer recommend). HelloWorld does indicate where to place the code to display a window; beyond that, it's probably easier to look at some real plugins for examples.

ClojureEditor, which is currently built, seems broken to me (the REPL doesn't evaluate, e.g., `(+ 1 1)`). Given that Arthur is no longer developing clooj and that few people use Clojure with Micro-Manager, I think it should be removed for now. (There is a newer version, 0.5, of clooj than the 0.2.7 used in the Micro-Manager plugin, which will become usable once we move to Java 8. When that happens, we could consider reviving ClojureEditor, if it seems helpful for debugging the acquisition engine.)

ChrisWeisiger, 2016-06-16:
ClojureEditor, DLLAutoReloader, Magellan, and TwoPhoton have been removed from the mm2 branch. They can still easily be recovered if necessary (by reverting the commit that removed them).

nicost, 2016-06-16:
and they should still be in the svn repository and the master branch on here on github.  Thanks!

It would be nice to add a very simple GUI to the HelloWorld example.  Would be plugin writers will often want to start with an example, and the other plugins all have their own styles (some of which are not great examples), so you can guide people in what you consider to be the right direction by providing a simple, but nice example.

closed 2016-08-10 by nicost