micro-manager/micro-manager #297: mm2 :  java.lang.NullPointerException at boot (Windows XP 32-bit)

hadim, 2016-06-16:
All is in the title. The log below : 

```
2016-06-16T09:47:37.687500 tid3568 [IFO,LogManager] Enabled primary log file C:\Program Files\Micro-Manager-2.0\CoreLogs\CoreLog20160616T094737_pid4316.txt
2016-06-16T09:47:37.687500 tid3568 [IFO,App] EDTHangLogger: Started monitoring of EDT hangs
                                   [       ] [heartbeat timeout = 4500 ms, hang check interval = 1000 ms]
2016-06-16T09:47:37.687500 tid3568 [IFO,App] User: brouhard
2016-06-16T09:47:37.687500 tid3568 [IFO,App] Host: Eliot
2016-06-16T09:47:37.687500 tid3568 [IFO,App] MM Studio version: 2.0.0-beta3 20160615
2016-06-16T09:47:37.687500 tid3568 [IFO,App] MMCore version 8.3.0
2016-06-16T09:47:37.687500 tid3568 [IFO,App] Device API version 67, Module API version 10
2016-06-16T09:47:37.687500 tid3568 [IFO,App] Operating System: Windows XP (x86) 5.1
2016-06-16T09:47:37.687500 tid3568 [IFO,App] JVM: Java HotSpot(TM) Client VM, version 1.6.0_31, 32-bit
2016-06-16T09:47:38.062500 tid4556 [IFO,App] Found plugin org.micromanager.acquiremultipleregions.AcquireMultipleRegions@17893aa
2016-06-16T09:47:38.093750 tid4556 [IFO,App] Found plugin org.micromanager.asidispim.ASIdiSPIM@10cadab
2016-06-16T09:47:38.093750 tid4556 [IFO,App] Found plugin ch.epfl.leb.autolase.AutoLasePlugin@4f62ae
2016-06-16T09:47:38.093750 tid4556 [IFO,App] Found plugin org.micromanager.clojureeditor.ClojureEditorPlugin@ebc433
2016-06-16T09:47:38.093750 tid4556 [IFO,App] Found plugin org.micromanager.CRISP.CRISP@9b4c27
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin com.asiimaging.CRISPv2.ASI_CRISP_V2@1930c4
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin org.micromanager.plugins.framecombiner.FrameCombinerPlugin@1f49068
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin org.micromanager.plugins.frameprocessor.FrameProcessorPlugin@12c8355
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin GaussianTrack_@dd38e4
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin org.micromanager.hcs.HCSPlugin@11dd618
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin org.micromanager.imageflipper.FlipperPlugin@e1d1b3
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin org.micromanager.intelligentacquisition.IntelligentAcquisition@17cff8d
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin com.imaging100x.tracker.TrackerPlugin@ef8c58
2016-06-16T09:47:38.109375 tid4556 [IFO,App] Found plugin org.micromanager.multicamera.MultiCamera@10ac17d
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.multichannelshading.MultiChannelShading@1a390ce
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.patternoverlay.PatternOverlayPlugin@1c943c5
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin edu.umassmed.pgfocus.pgFocus@44fbb4
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.pipelinesaver.SaverPlugin@feb435
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.pixelcalibrator.PixelCalibratorPlugin@12d509c
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.projector.ProjectorPlugin@1ceff2
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.recall.RecallPlugin@a8e6c7
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.plugins.sequencebuffermonitor.SequenceBufferMonitor@11c6e12
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.slideexplorer.SlideExplorerPlugin@1578c65
2016-06-16T09:47:38.125000 tid4556 [IFO,App] Found plugin org.micromanager.splitview.SplitView@10c81b4
2016-06-16T09:47:38.171875 tid4556 [IFO,App] Found plugin org.micromanager.autofocus.HardwareFocusExtender@1506a2b
2016-06-16T09:47:38.171875 tid4556 [IFO,App] Found plugin org.micromanager.autofocus.Autofocus@18d31e8
2016-06-16T09:47:38.171875 tid4556 [IFO,App] Found plugin org.micromanager.autofocus.AutofocusDuo@1aeee60
2016-06-16T09:47:38.171875 tid4556 [IFO,App] Found plugin org.micromanager.autofocus.AutofocusTB@b3333f
2016-06-16T09:47:38.187500 tid4556 [IFO,App] Found plugin org.micromanager.autofocus.OughtaFocus@1278ed4
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.acquisition.internal.DefaultMDAPlugin@d9bac7
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.display.internal.inspector.OverlaysPlugin@1d62a59
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.display.internal.overlays.ScaleBarPlugin@1764d7d
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.display.internal.overlays.TimestampPlugin@1d9797b
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.AutofocusButtons@95e588
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.CloseAllButton@75b220
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.ExposureTime@47c17a
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.LiveButton@103aa06
2016-06-16T09:47:38.375000 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.MDAButtons@17f5d54
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.PresetButton@14546f1
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.RefreshButton@1349389
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.SavedMDAButton@10418d2
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.ScriptButton@41e278
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.ShutterControl@17b70ef
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.SnapButton@81e6b6
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Found plugin org.micromanager.quickaccess.internal.controls.TextLabel@51b803
2016-06-16T09:47:38.390625 tid4556 [IFO,App] Plugin loading took 672ms
2016-06-16T09:47:38.500000 tid3568 [IFO,App] Finished waiting for plugins to load
2016-06-16T09:47:42.468750 tid3568 [IFO,App] 
                                   [       ] java.lang.NullPointerException in Thread[AWT-EventQueue-0,6,main]
                                   [       ]   at org.micromanager.internal.dialogs.IntroDlg.getUserName(IntroDlg.java:402)
                                   [       ]   at org.micromanager.internal.MMStudio.initializationSequence(MMStudio.java:418)
                                   [       ]   at org.micromanager.internal.MMStudio.<init>(MMStudio.java:317)
                                   [       ]   at MMStudioPlugin$1.run(MMStudioPlugin.java:82)
                                   [       ]   at java.awt.event.InvocationEvent.dispatch(Unknown Source)
                                   [       ]   at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
                                   [       ]   at java.awt.EventQueue.access$000(Unknown Source)
                                   [       ]   at java.awt.EventQueue$1.run(Unknown Source)
                                   [       ]   at java.awt.EventQueue$1.run(Unknown Source)
                                   [       ]   at java.security.AccessController.doPrivileged(Native Method)
                                   [       ]   at java.security.AccessControlContext$1.doIntersectionPrivilege(Unknown Source)
                                   [       ]   at java.awt.EventQueue.dispatchEvent(Unknown Source)
                                   [       ]   at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
                                   [       ]   at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
                                   [       ]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
                                   [       ]   at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
                                   [       ]   at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
                                   [       ]   at java.awt.EventDispatchThread.run(Unknown Source)

```

hadim, 2016-06-16:
I forgot to mention I am using the last build `MMSetup_32bit_2.0.0-beta3_20160615.exe` while `MMSetup_32bit_2.0.0-beta3_20160411.exe` works fine.

ChrisWeisiger, 2016-06-16:
This has been fixed. Thanks for the bug report!

closed 2016-06-16 by ChrisWeisiger

hadim, 2016-06-16:
Nice