micro-manager/micro-manager #302: HCS Plugin: Multi-point method for XY calibration would be useful

marktsuchida, 2016-06-17:
Finding the center of the (top-left) well is either hard, or time-consuming.

An alternative, 2-step procedure asking the user for the left and top edges might be easier to use.

Or a 4-step procedure asking for the top, left, bottom, right edges of the well might be more accurate for finding the correct center.

Suggested by Philippe: http://micro-manager.3463995.n2.nabble.com/HCS-plugin-more-accurate-x-y-stage-calibration-suggestion-tp7587195.html

nicost, 2016-06-22:
Agreed.  A more pressing issue for us is that it is often inconvenient to move to the center of well A1 (especially with high na objectives, since these may crash into the sides of the stage).  It would be very nice have an option to home on another well than A1 (it can even be argued that a well close to the center may be a better default option).

nicost, 2016-07-21:
I am working on all of this in https://github.com/nicost/micro-manager/tree/hcs

nicost, 2016-07-26:
Implemented in #369

closed 2016-12-04 by nicost