micro-manager/micro-manager #303: Zoom in/out icons placement

Philippe-P, 2016-06-20:
In MM2 live window, there are useful icons to zoom in/zoom out.  However, their positions is such that every time they are activated, their position on the screen changes. So if the user wants to  zoom in/out several times, she/he has to look for them, once localized move the mouse on top of them, and click.
Putting the icons in the upper left corner of the window would eliminate the problem.
It may not be a priority, but it sure would improve the heavy "in/out zoomer" experience.

ChrisWeisiger, 2016-08-03:
It'll probably take us a long time to come up with a new layout / access system for the image display windows that doesn't have this problem. In the meantime, I recommend using ImageJ's magnifying glass tool for repetitive zooming. Note that you can zoom out using right-click.

nicost, 2020-04-04:
They are now in the upper left corner, and the user can repeatedly click them without moving the mouse.

closed 2020-04-04 by nicost