micro-manager/micro-manager #304: mm2 : Position of the zoom window reset when switching channel

hadim, 2016-06-20:
How to reproduce : 

- Enable live mode
- Zoom to the center of the image (for example if you want to see something specific on the image)
- Switch channel
- The zoom level is still the same BUT the position of the zoom is set to the origin of the image (upper left part) which is not what I am excepted

Can you confirm this ?

ChrisWeisiger, 2016-06-20:
I can't confirm this using demo hardware. What kind of image dimensions and exposure times are you using?

hadim, 2016-06-20:
I test it again few minutes ago and I confirm the issue.

I am using mm2 build 20160615. I have the issue with both Andor iXon (512x512) and sCMOS camera (2560x2160).

I also confirm the demo config works well with the same MM installation.

My channel group only have two property `Core-Shutter` and `ZeissReflectorTurret-Label`.

Let me know if you need me to do more tests.

marktsuchida, 2016-06-25:
Just to be clear, is this with the zoom buttons ("+" and "-") in the image window, or are you changing the camera ROI?

hadim, 2016-06-25:
The zoom button in the image window.

nicost, 2020-04-04:
Since the display window is no longer re-created when switching channels, this is most likely fixed now (although I can not reproduce with the democamera, that apparently was previously not possible either, so please do re-open if not fixed).

closed 2020-04-04 by nicost