micro-manager/micro-manager #305: HCS, step 2 in nigh mode, available devices list looks ugly

nicost, 2016-06-22:
They are shown as black text on a (locally) white background.  Is it possible to show them as white text on a dark background, just as the top part of that window? It kinda looks bad the way it is....

ChrisWeisiger, 2016-06-22:
Could you please clarify what specific part of the GUI you're talking about? It's not obvious to me from the title. Looking at the HCS plugin, the only black text I see is the well coordinate indicators (column A, B, C, etc.). They should be fixed, but I don't think they're what you were talking about.

nicost, 2016-06-22:
I mean the names of the devices in the lower panel (next to the folder icons, and next to the dots when you "open" a folder). This is in Steps 2 of 7: Add or remove devices.

ChrisWeisiger, 2016-06-22:
Oh, you meant the HCW, not the HCS. :) Understood.

nicost, 2016-06-22:
Sorry!  My bad!

ChrisWeisiger, 2016-06-22:
This will require creating a custom renderer for the elements in the JTree that the devices display uses, which shouldn't be too complicated, but is nontrivial enough that I'm going let it take a backseat to other issues for now.

ChrisWeisiger, 2016-07-13:
This has now been fixed. Turns out no custom renderer was needed; instead there were look-and-feel colors that could be adjusted to achieve the desired effect. Said colors just weren't listed on the reference sheet I'd used when originally making the day/night system; they were however documented in `DefaultTreeCellRenderer`.

closed 2016-07-13 by ChrisWeisiger