micro-manager/micro-manager #309: ROI button no longer draws a centered square at half the image size

nicost, 2016-06-24:
which is a big pity, since it was the only way to quickly and easily set an ROI at the center quadrant of the camera.  Any way to get the old behavior back (or an improvement on the old behavior?)

marktsuchida, 2016-06-24:
Note that the "set ROI to center quad" is now a separate button (which does draw the selection), immediately to the right of the now "set ROI to selection" button.

The idea was to make it less confusing, because I've found myself explaining the complicated behavior of that single button over and over.

There should be an error message now if you click the "set ROI to selection" button without having a selection, so that users expecting the old behavior should not be confused.

nicost, 2016-06-24:
Perfect!  I simply did not notice that there are now 3 buttons instead of 2.

closed 2016-06-25 by marktsuchida