micro-manager/micro-manager #313: mm2: Various text are close to unreadable in night mode

nicost, 2016-06-25:
There are multiples of these, the most important are the read-only variables in the device/property browser and its derivatives (essential for me in pixel size calibrator), and the number next to the slider in the front panel in edit mode.

ChrisWeisiger, 2016-06-28:
This should be substantially improved now. Can we close this issue, or are there areas that remain hard to read?

ChrisWeisiger, 2016-07-07:
I'm going to assume this is now resolved. Please re-open if there are any extant hard-to-read texts.

nicost, 2016-07-13:
The number next to the slider is still unreadable when the group is selected (which happens after moving the slider and letting go of it).

ChrisWeisiger, 2016-07-14:
Okay, this should now be fixed. Closing again; please re-open if you find another place where text is hard to read.

closed 2016-07-14 by ChrisWeisiger