micro-manager/micro-manager #314: mm2: Suggestion for the device/property browser

nicost, 2016-06-25:
The device/property browser and related tables are difficult to work with since they contain so many entries.  It is often hard to find the property that I am looking for.  Chris Baumann suggested to organize the devices in tabs, so that you click on a tab and then only see the properties for that device.  That makes a lot of sense to me, and would make things more pleasant.

jondaniels, 2016-06-25:
I like this idea.  A related one (not mutually exclusive) is to add a search/filter box where you can type part of the property name and it will only show matching property names.  Usually I know at least part of the name of the property but indeed it can be tedious to find.

ChrisWeisiger, 2016-06-29:
How about a dropdown menu that lets you select the device to view properties for (with an "All Devices" default selection)? That would be easier to implement since it would work similarly to the existing checkboxes, and it also would scale better to setups that have large numbers of devices -- I worry about a GUI that would have 10+ tabs, for example.

I like the text box for searching by name, too.

nicost, 2016-06-29:
For the user I think that tabs would be slightly nicer than a dropdown, but I can see the implementation pain and a dropdown will already be a great improvement.

jondaniels, 2016-06-30:
If you did want to use the tab idea, there could be a tab for each category of devices.  I.e. the equivalent of selecting only one of the existing checkboxes at a time.

marktsuchida, 2016-07-01:
I agree with @ChrisWeisiger that tabs would be problematic (e.g. `NIMultiAnalog` can create up to 32 `ao` ports, each a device). Even if mitigated by e.g. grouping by hub, I find it unlikely that it will work well given the limited width available.

I feel that the search/filter box is the first thing that should be added and cannot go wrong.

A popup could be useful if it includes both individual devices and device categories, but won't replace some of the ways in which the checkboxes can currently be used (I often like to view shutters and state devices together, for example). I think we need more ideas here.

Tabs with the categories has the same issue (can't view two categories together).

Here is one hybrid idea (I'm not married to it):
- Keep the category checkboxes
- Add a search/filter box
- Replace the checkbox labels with popup menus, whose items are the devices in each category
- When a popup item (a device) is selected, it gets entered into the search/filter box, thus resulting in only that device being shown (the corresponding checkbox is checked if not already checked, but other checkboxes are not affected)

Cons:
- Complicated
- The device label could be a partial match to other devices (e.g. "Camera" matches "Camera2"), so that undesired devices may be shown after selecting from the popup menus

Yet another idea (with the goal being making it easier to find a particular device) would be to include prominent headers for each device (1.5-2x the height of property rows) in the table.

jondaniels, 2016-07-01:
I like Mark's idea, including the separate one of putting extra rows in the
table.

I think the partial match problem can be mitigated by adding a dash to the
end of the device name when adding the device name to the search box (that
is the delimiter used in the property browser between device name and
property name).

In case it wasn't clear the search should also work on property names, e.g.
if you search for "trigger" you may get properties from stages, cameras,
and shutters and probably more than one from some devices.

marktsuchida, 2016-07-01:
@jondaniels Good point, I like the trailing dash idea (even if it's not 100%).

ChrisWeisiger, 2016-07-13:
#339 adds a search text field to the Device Property Browser and Config Group Editor. You can do a case-insensitive search for device names, property names, or a combination (using a dash to delimit between the device and property name portions of your query). Let's give this change a shot; I think it's pretty effective. It may be that we don't need to add any more ways to filter the view.

nicost, 2016-07-18:
I love this!  Works very well.

closed 2016-08-22 by nicost