micro-manager/micro-manager #320: mm2: Metadata class has TODO items

marktsuchida, 2016-06-30:
`getPixelAspect()` - this is always set to `1.0` by acqEngine, which is a lie. Let's remove it from acqEngine. Okay to keep in `Metadata` but we should not set it until we have a mechanism to know the aspect ratio.

`getStartTimeMs()` - this does not make sense in per-plane metadata. It would make sense in summary metadata, where it always has been (now named `getStartDate()`). Delete.

`getInitialPositionList()` - only makes sense in summary metadata (which already has `getStagePositions()`). Delete.

`getPixelType()` - clearly it doesn't make sense for this to be settable independently of `ijType`. Make the implementation (`DefaultMetadata`) return the string version of `getIjType()` (and also map builder method)?

`getSource()` - this is set to `"Micro-Manager"`. Given that summary metadata already includes Micro-Manager version, probably best to delete.

Also `SummaryMetadata.getPrefix()` and `SummaryMetadata.getDirectory()` - what is the difference?

ChrisWeisiger, 2016-06-30:
I went ahead and removed the startTimeMs, initialPositionList, pixelType, and source properties from the `Metadata` class. I thought about effectively aliasing pixelType to ijType, but considering that neither value is currently read by any code (that we have, anyway), the only effect is removing an entry from the MetadataPanel -- and we have other places in the GUI where we display the pixel type in a more readable fashion. And code-wise, you should rely on what the `Image` says its pixel type is, not what the `Metadata` claims it to be. So it doesn't seem vital to keep a more human-readable form of the pixel type in the `Metadata` class.

Regarding the `prefix` and `directory` fields: the latter is the directory that contains the folder that contains the dataset, while the former is the name of the folder that directly contains the TIFFs. For example, `directory` could be `/Users/chriswei/data` and `prefix` could be `TestAcq_1`, indicating that `/Users/chriswei/data/TestAcq_1` contains an MM dataset. It's entirely possible there's a better way to handle this; the directory/prefix split was, if I recall correctly, basically just copied over from the 1.4 summary metadata unchanged. Note that if you want to get the location of a dataset on disk, `Datastore.getSavePath()` should be authoritative...and it references neither of these values. Instead, it's a purely-in-memory variable, which is set by `StorageMultipageTiff` and `StorageSinglePlaneTiffSeries` in their constructors. This setup probably needs to be changed so it enforces internal consistency somehow.

marktsuchida, 2016-07-01:
Now I see `"Source"` showing up in `userData` -- except that its value appears to be the camera name rather than `"Micro-Manager"`...

I agree that _both_ `pixelType` and `ijType` should be removed from `Metadata`, given that `Image` already contains that information. (Of course, these values should still be written to files for compatibility if they were written in 1.4, but their values should be computed from `Image`.)

Prefix vs directory seems fine, thanks for the clarification (perhaps javadoc can be updated). I got confused since the word "directory" seems to mean "directory + prefix" in other parts of the API (e.g. parameter to `DataManager.createMultipageTIFFDatastore()`; `DataManager.getUniqueSaveDirectory()`). Probably just a matter of documentation.

Any thoughts on `PixelAspect`?

ChrisWeisiger, 2016-07-01:
`keepShutterOpen[Channels|Slices]` should be moved to `SummaryMetadata`. Agreed with Mark that we should remove `ijType` from the `Metadata`.

ChrisWeisiger, 2016-07-05:
The "keep shutter open" properties have been moved to `SummaryMetadata`, and the `ijType` field has been removed from `Metadata`. I believe that's everything.

closed 2016-07-05 by ChrisWeisiger