micro-manager/micro-manager #323: Exposure time not updating in multi - D Acqu.

impact27, 2016-07-04:
When taking a multi - D acquisition, if the exposure time is changed, the image is modified but the metadata is not.

Eg: I start a multi D acqu. with 10ms exposure time, go to 100ms exposure time during the acquisition, but the image metadata is unchanged. 

MicroManager Version 1.4.21

ChrisWeisiger, 2016-07-04:
Unfortunately, there's not really anything we can do about this with the current acquisition engine. It's not expected that the user make changes to the hardware while the acquisition engine is in operation, so we don't "react" to such changes as far as the metadata is concerned. This is necessary for performance reasons; if we were continually checking for changed hardware properties, then acquisitions would take a lot longer.

marktsuchida, 2016-07-21:
But this should eventually be fixed... so reopening.