micro-manager/micro-manager #324: Script panel does not "see" classes loaded as plugins

nicost, 2016-07-07:
For instance, the following Beanshell script:
```
import org.micromanager.projector.ProjectorControlForm;
pcf = ProjectorControlForm.showSingleton(core, mm);
```
Generates an error when run in Micro-Manager 2.0, even when the Projector plugin is open.  It does not generate an error when run in Netbeans (when the Projector plugin is part of the project).  I tried adding the Projector.jar directly to the classpath, but that leads to an error on startup.  I guess this currently means that scripts can not call plugins, which is significant problem for us.

ChrisWeisiger, 2016-07-07:
This was intentional: we did not want to pollute the main program's namespace with plugin classes, as a matter of principle (otherwise a plugin could insert items into the `org.micromanager` namespace or do other harmful things).

You should still be able to access the plugin in a more general fashion by using the `PluginManager`: `mm.plugins().getMenuPlugins().get("org.micromanager.projector.ProjectorPlugin")` will get you the `MenuPlugin` instance corresponding to the Projector plugin, for example. 

What error do you get on startup if you add Projector.jar to the classpath?

nicost, 2016-07-07:
Is this documented somewhere?

Putting the Projector.jar on the classpath leads to a bunch of errors.  
Easy enough to try if you want to trouble shoot.

On 7/7/16 7:28 AM, ChrisWeisiger wrote:
> This was intentional: we did not want to pollute the main program's
> namespace with plugin classes, as a matter of principle (otherwise a
> plugin could insert items into the |org.micromanager| [...]

ChrisWeisiger, 2016-07-07:
The PluginManager is documented in our JavaDocs:
http://valelab.ucsf.edu/~MM/doc-2.0.0-beta/mmstudio/org/micromanager/PluginManager.html
which should be readily discoverable when examining the documentation for the `Studio` class.

The change to how plugins are loaded is not documented so far as I'm aware, because it was expected to be an implementation detail.

As for the classpath issue, I modified the mmimagej launcher by adding

`-classpath "mmplugins/Projector.jar" \

immediately prior to the `-eval` line, and was able to import and instantiate the ProjectorPlugin with no difficulties.

nicost, 2016-07-07:
> The PluginManager is documented in our JavaDocs:
> http://valelab.ucsf.edu/~MM/doc-2.0.0-beta/mmstudio/org/micromanager/PluginManager.html [...]

That is hard/impossible to find out for any non-full-time programmer.  
Can this be added to the Version 2.0 API page or the "API, How do I" 
page?  An example script would also be nice.

> The change to how plugins are loaded is not documented so far as I'm
> aware, because it was expected to be an implementation detail.
>
> As for the classpath issue, I modified the mmimagej launcher by [...]
Funny.

Try:

```
java -cp ij.jar:mmplugins/Projector.jar ij.ImageJ
Asked to load file at path 
/Applications/Micro-Manager2.0/GlobalUserProfile.txt which doesn't 
exist; instead providing a default empty PropertyMap.
2016-07-07T11:26:31.420535 tid0x700001762000 [IFO,LogManager] Enabled 
logging to stderr
2016-07-07T11:26:31.588563 tid0x700001762000 [IFO,LogManager] Enabled 
primary log file 
/Applications/Micro-Manager2.0/CoreLogs/CoreLog20160707T112631_pid3021.txt
2016-07-07T11:26:31.594600 tid0x700001762000 [IFO,App] EDTHangLogger: 
Started monitoring of EDT hangs
                                              [       ] [heartbeat 
timeout = 4500 ms, hang check interval = 1000 ms]
2016-07-07T11:26:31.594658 tid0x700001762000 [IFO,App] User: nico
2016-07-07T11:26:31.594754 tid0x700001762000 [IFO,App] Host: 
valenico.ucsf.edu
2016-07-07T11:26:31.594766 tid0x700001762000 [IFO,App] MM Studio 
version: 2.0.x dev
2016-07-07T11:26:31.594817 tid0x700001762000 [IFO,App] MMCore version 8.3.0
2016-07-07T11:26:31.594843 tid0x700001762000 [IFO,App] Device API 
version 67, Module API version 10
2016-07-07T11:26:31.594858 tid0x700001762000 [IFO,App] Operating System: 
Mac OS X (x86_64) 10.11.5
2016-07-07T11:26:31.594872 tid0x700001762000 [IFO,App] JVM: Java 
HotSpot(TM) 64-Bit Server VM, version 1.8.0_66, 64-bit
2016-07-07T11:26:32.125863 tid0x700001aee000 [IFO,App] Unable to load 
JAR at file:/Applications/Micro-Manager2.0/mmplugins/Projector.jar
                                              [       ] 
java.lang.NoClassDefFoundError: org/micromanager/MenuPlugin in 
Thread[Plugin loading thread,6,main]
                                              [       ]   at 
java.lang.ClassLoader.defineClass1(Native Method)
                                              [       ]   at 
java.lang.ClassLoader.defineClass(ClassLoader.java:760)
                                              [       ]   at 
java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
                                              [       ]   at 
java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
                                              [       ]   at 
java.net.URLClassLoader.access$100(URLClassLoader.java:73)
                                              [       ]   at 
java.net.URLClassLoader$1.run(URLClassLoader.java:368)
                                              [       ]   at 
java.net.URLClassLoader$1.run(URLClassLoader.java:362)
                                              [       ]   at 
java.security.AccessController.doPrivileged(Native Method)
                                              [       ]   at 
java.net.URLClassLoader.findClass(URLClassLoader.java:361)
                                              [       ]   at 
java.lang.ClassLoader.loadClass(ClassLoader.java:424)
                                              [       ]   at 
sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
                                              [       ]   at 
java.lang.ClassLoader.loadClass(ClassLoader.java:411)
[... 305 lines cut ...]
java.awt.EventQueue.access$500(EventQueue.java:97)
                                              [       ]   at 
java.awt.EventQueue$3.run(EventQueue.java:709)
                                              [       ]   at 
java.awt.EventQueue$3.run(EventQueue.java:703)
                                              [       ]   at 
java.security.AccessController.doPrivileged(Native Method)
                                              [       ]   at 
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
                                              [       ]   at 
java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
                                              [       ]   at 
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
                                              [       ]   at 
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
                                              [       ]   at 
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
                                              [       ]   at 
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
                                              [       ]   at 
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
                                              [       ]   at 
java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

marktsuchida, 2016-07-07:
> The PluginManager is documented in our JavaDocs:
>  http://valelab.ucsf.edu/~MM/doc-2.0.0-beta/mmstudio/org/micromanager/PluginManager.html [...]

> That is hard/impossible to find out for any non-full-time programmer.
> Can this be added to the Version 2.0 API page or the "API, How do I"
> page?  An example script would also be nice.

I agree, this will be needed for things like the ASIdiSPIM API.
Please feel free to add to the wiki how-to page!

    mm.plugins().getMenuPlugins().get("org.micromanager.projector.ProjectorPlugin").onPluginSelected();

ChrisWeisiger, 2016-07-07:
Regarding the classpath issue, you need to add a `-jar ij.jar` after specifying the classpath, e.g. `java -cp ij.jar:mmplugins/Projector.jar -jar ij.jar ij.ImageJ`.

nicost, 2016-07-08:
```
core = mm.getCMMCore();
ProjectorControlForm = mm.plugins().getMenuPlugins().get("org.micromanager.projector.ProjectorControlForm");
psf = ProjectorControlForm.showSingleton(core, mm);
```
results in:
```
 run-time error : Null Pointer in Method Invocation
```
How do I call a static function in a class loaded through mm.plugins().getMenuPlugins() ?

marktsuchida, 2016-07-08:
@nicost `getMenuPlugins()` will give you the menu plugin _instances_. The `ProjectorControlForm` is not a menu plugin (just a random internal class of the Projector plugin), so it cannot be accessed this way.

For something like that to be possible, `ProjectorPlugin` would have to provide a (non-static) API method.

What is the motivation here? Are you simply trying to open the Projector plugin from a script? What's wrong with the `onPluginSelected()` method?

nicost, 2016-07-08:
> @nicost <https://github.com/nicost> |getMenuPlugins()| will give you
> the menu plugin /instances/. The |ProjectorControlForm| is not a menu
> plugin (just a random internal class of the Projector [...]

So, I can add a non-static function to ProjectorPlugin that will return 
the singleton instance of ProjectorControlForm?

> What is the motivation here? Are you simply trying to open the 
> Projector plugin from a script? What's wrong with the 
> |onPluginSelected()| method?
>

I need to get a singleton instance of the plugin (or in this case, the 
ProjectorControlForm) from a script so that I can programmatically use 
that singleton instance.  How does one use the ```onPluginSelected()`` 
method to get that?  I do not see that function return an instance.

Functionality like this is quite important for any script or plugin that 
wants to make use of other plugins.  This used to be quite easy when 
everything was on the class path.  I understand the desire to avoid 
plugins putting random stuff on the class path, but there needs to be a 
clear and simple example that demonstrates how other plugins and scripts 
can use plugin code.

marktsuchida, 2016-07-08:
Do you mean as a temporary hack, or as a permanent solution?

Perhaps I'm stating the obvious, but plugin code should not normally be considered part of any API (it's like `org.micromanager.internal.*`), and surely `ProjectorControlForm` could be refactored without notice.

If we want other code to be able to reuse aspects of the Projector plugin, then the Projector plugin will need to be given some sort of API. Which is why I'm asking what you are trying to do with `ProjectorControlForm` other than just displaying the window (what do you mean by "programmatically use"?). I was confused because `ProjectorControlForm.showSingleton()` (which is what your initial question was about) doesn't do anything more than what `ProjectorPlugin.onPluginSelected()` already does.

Actually, the fact that there is a static method to get the singleton `ProjectorControlForm` is likely a historic relic, because in mm2 it is ensured that a unique instance of `ProjectorPlugin` is available (from which a unique instance of `ProjectorControlForm` could easily be managed). In any case, it is technically possible to add a method to `ProjectorPlugin` that returns the `ProjectorControlForm` singleton instance, but `ProjectorControlForm` is not suitable to be considered an API in its current form. I very much agree that having an API for some of Projector's functionality would be great, but it just isn't that way at the moment (this is technical debt, of course).

P.S. More generally, when there is a plugin that should be usable as a library, I think the best thing to do is to create a separate library JAR (`plugins/Micro-Manager/Foo.jar`), and make the plugin JAR (`mmplugins/FooPlugin.jar`) a (possibly thin) wrapper around it. That way, Java code can import the library classes. Alternatively, `FooPlugin.jar` could contain all the code, and `Foo.jar` (or `Foo-api.jar`) could contain just the interface definitions for the API. I'll add this to the 2.0 plugins wiki page.

ChrisWeisiger, 2016-07-08:
I agree with Mark's suggestion. The "plugin API" jar would of course need to be installed into $APPDIR/plugins/Micro-Manager (as opposed to $APPDIR/mmplugins, which is where normal plugin jar files go) if it is to be accessible by other code. That ought to be a fairly straightforward modification of the build scripts, though.

nicost, 2016-07-08:
I think that it will be highly preferable to stop relying on the ImageJ class loader to load jars from ```plugins/Micro-Manager```, but create a ```jars``` directory and put its content on the class path of the jvm (exactly the way Fiji does).  This approach is for instance needed to get the ClearVolume plugin to work, and will reduce similar problems in the future.  I think that the only jar that needs to be in the ImageJ plugins directory is ```MMJ_.jar```.

marktsuchida, 2016-07-08:
I generally agree that JARs in `plugins/Micro-Manager` (but not those in `mmplugins`) should be moved to `jars`, although I think that is a separate issue.

Unless you mean to also move mmplugin JARs into `jars`, and completely replace the plugin loading mechanism with a classpath-based one. I'm not against that in principle, though I would like to see the plugin Java packages renamed first, so that non-API code is ensured to be in `internal` packages. Other than that, I don't immediately see any issue. (However, I'm not sure I'm recalling all the downsides of this approach we carefully considered when deciding how to load plugins in mm2. -- Among other things, startup performance was tested, IIRC.)

nicost, 2016-07-09:
> I generally agree that JARs in |plugins/Micro-Manager| (but not those 
> in |mmplugins|) should be moved to |jars|, although I think that is a 
> separate issue.
>
> Unless you mean to also move mmplugin JARs into |jars|, and completely 
> replace the plugin loading mechanism with a classpath-based one.
>

No, that it is not what I meant.  Relying on the ImageJ class loader to 
load tons of jars will lead to problems (as we already experience with 
the ClearVolume plugin) and it will be much important to have those 
directly on the classpath.

marktsuchida, 2016-07-09:
I hope #330 addresses the original issue.

closed 2016-07-09 by marktsuchida