micro-manager/micro-manager #327: Compatibility with USB3Vision

derinsevenler, 2016-07-08:
To the best of my knowledge, Micro-manager 1.4.xx does not currently support USB3Vision cameras.

I have developed an adapter for USB3Vision cameras, which I have been using in our lab for about six months: https://github.com/derinsevenler/usb3VisionCamera 

It is based on the existing GigECamera adapter: https://micro-manager.org/wiki/GigECamera.
The GigECamera adapter is built to interface with an older version of the JAI SDK. This USB3Vision adapter has small api-related changes which allow it to interface with a newer version of the same JAI SDK (version 2.1.6) which also has support for USB3Vision-compliant cameras. It seems possible that this adapter could work for both USB3Vision and GigEVision compliant cameras, but this is untested speculation.

This is my first time contributing to an open-source project so I am looking for your guidance in how I should format & document my contribution. Thanks everyone!

marktsuchida, 2016-07-08:
Sounds great! I know a bunch of people have wanted this. If you have your code on a Git branch, you can create a pull request. Otherwise, please contact us offline.

The existing GigECamera code doesn't follow the code formatting standards of the rest of the project, but it is more practical to follow its local convention than to mix styles (unless reformatting the whole device adapter). Documentation for users will live on our wiki, once the device adapter has been added.

derinsevenler, 2016-07-08:
Thanks Mark! I believe I've created a branch and submitted a Pull Request to the master branch. I guess further discussion should take place on there, so maybe this issue is closed?

marktsuchida, 2016-07-08:
Thanks (#328).

closed 2016-07-08 by marktsuchida