micro-manager/micro-manager #329: DisplayWindow.getImagePlus regularly returns null when there is no apparent reason to do so

nicost, 2016-07-08:
This problem can be observed using the Beanshell script:
```Beanshell
mm.live().snap(true);
ip = mm.live().getDisplay().getImagePlus();

if (ip == null) {
	mm.scripter().message("This time no imagePlus");
}
else {
	mm.scripter().message("Found an ImagePlus");
}
```
Immediately after startup and when applying an ROI, snapping, pressing the button with th green arrows on all corners and running vthe script again reliably trigger the output "This time no ImagePlus";

Maybe related, is there a function to create an ImageJ ImagePlus object from a Micro-Manager Image object?

ChrisWeisiger, 2016-07-08:
I believe what is going on here is that the call to getImagePlus() occurs before the display has been recreated (the display must be recreated whenever the image dimensions change, e.g. due to a change in ROI). The SnapLiveManager.snap() method does not block until the display is recreated; it just gets the images from the Core, starts the process of inserting them into the snap/live Datastore, and returns the raw images. I say "starts the process" because if processors are involved, then it make take a long time for images to reach the Datastore and thus be displayable. Indeed, in the case of, say, a frame-averaging plugin that has not accumulated enough frames to show an averaged frame, it could take an arbitrary amount of time for any images to be shown.

Since the display derives its parameters, including the ImagePlus, from the images in the Datastore, it cannot create the ImagePlus until after it has at least one image to work with; until that time it returns null. 

Of course that's highly non-ideal for your use case. I suppose we could have a method on DisplayWindow that blocks until the DisplayWindow has actually drawn something to screen. The alternative would be to have your code listen for the DisplayAboutToShow event and check if the display that is nearly visible is in fact the Snap/Live display, but that sounds rather clunky.

nicost, 2016-07-09:
Yes, it would be very clunky to have to add event listeners to scripts.  It would be much nicer to be able to generate an ImagePlus directly from the image returned by the SnapLiveManager.snap() method.  I guess that I could write that method myself, but it seems that this would be a very commonly used scenario.  Is there already a method somewhere in the code that carries out that conversion that could be called as a utility function?

On the other hand, we also would like to use the ImageProcessor pipeline, so I think that rules out using the result of the snap() command directly.  So, how should a script or plugin go about snapping an image and getting its hands on the processed resulting image (preferably as an ImagePlus, but also OK as a Micro-Manager Image and having a conversion function somewhere)?

ChrisWeisiger, 2016-07-09:
Oh right, sorry. I meant to answer your second question but got distracted. Look at mm.data().ij(), the ImageJConverter:
http://valelab.ucsf.edu/~MM/doc-2.0.0-beta/mmstudio/org/micromanager/data/ImageJConverter.html

It can create an ImageProcessor from an MM Image object; you should be able to create a new ImagePlus from that if you need to.

ChrisWeisiger, 2016-07-11:
I've added `DisplayWindow.waitUntilVisible()`, which you can call after calling snap() to wait until the DisplayWindow has content to display; at that point, it will provide a valid ImagePlus. I think that covers everything for this issue.

closed 2016-07-11 by ChrisWeisiger

nicost, 2016-07-11:
That sounds great, thanks!  Is it guaranteed that the DisplayWindow will 
contain the image that was just snapped when one calls waitUntilVisible()?

ChrisWeisiger, 2016-07-11:
No, it is not. The application image processing pipeline stands between the snap() call and whatever is displayed in the DisplayWindow, so theoretically anything could be displayed. Even if you have no processing pipeline, there's still no guarantee, as it is theoretically possible for another thread to call `SnapLiveManager.displayImage()` in-between the call to snap() and the creation of the DisplayWindow.

However, in practice if there's no processing pipeline, you can reasonably assume that the image(s) in the DisplayWindow will match the return value of calling snap() (i.e. the raw images from the Core).

nicost, 2016-07-11:
And how does one make sure to get the correct image when there is a 
processing pipeline?  Say, I snap an image, have a pipeline to do 
flatfield correction and want to analyze the flatfield corrected image?

ChrisWeisiger, 2016-07-11:
Call `DisplayWindow.getDisplayedImages()` once the display becomes visible.

nicost, 2016-07-11:
For anyone who finds this and also wants to snap an image, have it processed by the default pipeline, and then analyze it using ImageJ code, I am now using the following Beanshell script:

```
import ij.ImagePlus;

store = mm.data().createRAMDatastore();
dw = mm.getDisplayManager().createDisplay(store);  // only needed if display is desired
pipeLine = mm.data().copyApplicationPipeline(store, true);  // pipeline insertion will block
imgList = mm.acquisitions().snap();
coords = mm.data().createCoords("t=0,p=0,c=0,z=0");
img = imgList.get(0).copyAtCoords(coords);
pipeLine.insertImage(img);
processedImg = store.getImage(coords);
iProcessor = mm.data().ij().createProcessor(processedImg);
ip = (new ImagePlus("tmp", iProcessor)).duplicate();   // duplicate if you want to store the original images
```

ChrisWeisiger, 2016-07-11:
Looks great! I encourage you to add it to https://micro-manager.org/wiki/Version_2.0_Scripts . :)

nicost, 2016-07-11:
Done.