micro-manager/micro-manager #338: Binning and ROI are not reliably present in the metadata

nicost, 2016-07-13:
When using the demo-camera, both when snapping an image (using the Snap button in the GUI), and when running an MDA (by unchecking all check-boxes so that the MDA takes only a single image), both the "Binning" and "ROI" field are correctly populated in the Image metadata.

However, when doing the same with the Andor Zyla camera (SDK3 driver), the behavior is different.  Upon snapping, there is a "Binning" property in the metadata, but no ROI unless a ROI is set (i.e. the behavior is different from the demo camera, which shows a ROI in the metadata even when no ROI is set).  When running an MDA, the ROI property is present, but the Binning property is not.  When the ROI is set through a script, rather than the UI, the ROI property is not in the metadata.

With a Hamamatsu camera, both ROI and Binning are present when snapping (even when ROI is set), but Binning is not set when running an MDA. Even when the ROI is set through a script it is present in the metadata.

The most important problem is that the Multi-Shading plugin is looking at the Binning and ROI properties of the metadata, to match the flat-field based on the current Binning and ROI settings.  When the Binning property is not present is assumes that the binning is 1, and when the ROI property is not present, it assumes that the camera is set to be full frame.  Clearly, those assumptions can be very wrong.  The alternative for the multi-shading plugin would be to do nothing when either the Binning or ROI metadata are null, but that would in practice mean that the Multi-Shading plugin is useless.  If you can give me clues where to start looking (I guess this is something in the device adapters?) then I'll be happy to do some digging as this directly interferes with our project (that relies on the Multi-Shading plugin working reliably).

ChrisWeisiger, 2016-07-13:
Unfortunately, this appears to be complicated. Here's the following sources I could find for the Binning and ROI metadata fields (taken from MM2.0):

* The device adapter. If you look at DemoCamera's `InsertImage()`, it creates a `Metadata` object (NB not the Java object of course) and inserts the `MM::g_Keyword_Metadata_ROI_X` and `MM::g_Keyword_Metadata_ROI_Y`, and `MM::g_Keyword_Binning` metadata fields. 

* MMCoreJ.i: its `createTaggedImage()` method adds ROI and Binning fields to the `TaggedImage`'s tag's (though the binning field is in a try/catch, which implies it may be prone to failure). I'm not certain what happens to the values provided by the device adapter at this point.

* The acquisition engine. Its `generate-metadata` method also creates a "Binning" metadata field, but puts the ROI metadata in the ''summary'' metadata (in `make-summary-metadata`).

* The `AcquisitionManager`'s `generateMetadata()` method provides a Binning property but no ROI property. This method is used by the Snap/Live manager.

The latter two items should be easy to fix, which should help some. I'm at a loss for why doing things via a script would make a difference vs. doing them from the GUI, though, unless the problem is that the system state cache was not updated after setting the ROI. You might try calling `mmc.updateSystemStateCache()` after setting the ROI in your script.

nicost, 2016-07-13:
I assume (but not 100% sure), that both snapping using the snap button and running an MDA using a single image will "snap" and not use a sequence acquisition.  The device adapter's ``InsertImage()``` function will only get into play when running a sequence acquisition.  So, I think that for the situations I describe above only the last 3 of your items are relevant (although it would be good to ascertain that sequence acquisitions also provide the correct metadata).  

Are there easy ways to reduce the number of code paths generating metadata, so that users can be sure of at least a small set of reliable metadata (binning, exposure, ROI are certainly part of that set)?  In the meat time it would be great to see what fixes to the last two items you describe will do for the real cameras.

nicost, 2016-07-13:
Just tested with the DemoCamera and found that in all situations (snapping using the GUI, MDA with single image, MDA with sequence) the code path goes through the ```MMCoreJ.i``` derived ```createTaggedImage(pixels, md)``` function, which is the place where the ROI and Binning properties are added.  It may be a good idea to remove all duplication of this code in the acquisition engine and the ```AcquisitionManager's createMetadata ``` code to avoid future confusion.  I'll try to debug this on real hardware.  One issue could be that the information is retrieved from the actual hardware rather than cache, and the cameras may not be in a mood to provide that information while snapping an image.

nicost, 2016-07-13:
Some more digging:

* On a second system, the Andor Zyla behaves the same as the Hamamatsu, i.e when snapping both Binning and ROI are set, when running an MDA, ROI is set, but Binning is not (I'll go back to the first system to figure out what is going on there).
* When getting the binning property from the camera, both the Andor and Hamamatsu report "1x1", rather than "1" as the demo camera does. ```createTaggedImage(Object pixels, Metadata md)``` in CMMCore.java inserts both the ROI and binning into the metadata, but the Binning property is later removed by some entity (I suspect the acquisition engine as that is the only thing that has access to metadata that I can not get to, although I do not understand how the property could disappear).  

Not sure how to fix this.  Both the acquisition engine and CMMCore insert ```BitDepth, PixelSizeUm, Width, Height, PixelType, Binning```, and some other properties which seems superfluous.  I am tempted to add code to MMCoreJ.i to change the String "1x1" to "1" (as ```generateMetadata``` in ```DefaultAcquisitionManager`` does).  It also confuses me why the camera devices have a GetBinning and SetBinning function, but the only way to get binning from userspace is through the binning property (I think that is the main design flaw that lead to the problem).

ChrisWeisiger, 2016-07-13:
My reading of the acquisition engine is that it is supposed to only overwrite the metadata that comes out of the Core if it has a non-nil replacement for it:

    (defn annotate-image [img event state elapsed-time-ms]
      {:pix (:pix img)
       :tags
       (merge-with #(or %2 %1) ; only overwrite tags if generated tag is not nil
         (:tags img)
         (generate-metadata event state)
         {"StateCache-keys" (JSONArray. (keys (:system-state state)))}
         {"ElapsedTime-ms" elapsed-time-ms}
         )}) ;; include any existing metadata

`generate-metadata` creates a `Binning` property but not an `ROI` property. And you're seeing the `Binning` property from the Core get overwritten in MDA, right? But `ROI` makes it through? It's possible that the acquisition engine is somehow faulty and thus deleting the provided-by-Core `Binning` property while not replacing it with a valid alternative. 

As an experiment, you could try deleting the "Binning" line from `generate-metadata` (just delete the entire line, it should be perfectly safe) to see if that causes the Core property to make it through. The acquisition engine's source for that property is calling `core.getProperty(core.getCameraDevice(), "Binning")`, which ought to be accurate, but it could potentially be going off the rails somewhere in the rest of the engine.

nicost, 2016-07-13:
Do you have instructions on how to run a modified acquisition engine?  Deleting a line is one thing, but how do I get that modified code to execute?

ChrisWeisiger, 2016-07-13:
You'd need to recompile the acquisition engine; it's compiled by Ant. I don't know how to set that up in NetBeans, though.

I've emailed you a jar of the compiled acquisition engine with no Binning property set.

nicost, 2016-07-14:
I think that I figured out at least part of the problem.  It is certainly caused by cameras returning String properties that contain something like "1x1" for Binning rather than Integer properties.  When ```DefaultMetadata.legacyFromJSON(tags)``` translates the JSON tags to the new system, it throws an exception in the line ```builder.binning(MDUtils.getBinning(tags));``` because ```MDUtils.getBinning(tags)``` only work with integer properties.  I changed that code so that it will translate Strings correctly, but for reasons beyond me, NetBeans no longer executes my code but rather executes the existing code, hence I am unable to check this without building binaries (which I am not setup to do on Windows).  

Even on System #1 I now see the ROI property most (all?) the time.  Not sure what is going on with that, but will look at that more later.

nicost, 2016-07-14:
Managed to test the code on System #1 and it does indeed fix the problem with the Binning property not propagating.  Also found out what is going on with the ROI: Occasionally, the camera decides that the ROI starts at pixel "-1" (I do not understand how we can even get such a value since the API for the Device Adapter's ```GetROI`` function uses unsigned integers) .  That results in a ROI String that looks like  "49--1-540-540".  When this String later on in the code is split, it results in an array with 5 members rather than 4, and the ROI will not be added to the metadata.  

There are many different ways of dealing with this, all with their advantages and disadvantages.  Interestingly, even setting the ROI to -1 appears to work as the following shows:
```
bsh % ROI = mm.getCMMCore().getROI();
<java.awt.Rectangle[x=49,y=-1,width=540,height=540]>
 mm.getCMMCore().setROI(ROI.x, ROI.y, ROI.width, ROI.height);
bsh % p = mm.getCMMCore().getROI();
<java.awt.Rectangle[x=49,y=-1,width=540,height=540]>
```
Maybe it is best to make the code in MDUtils.getROI(tags) more intelligent so that it interprets "--1" as a negative number?

ChrisWeisiger, 2016-07-14:
My inclination would be to modify `MDUtils.getROI()` to use a different delimiter than `-` when it converts rectangles to/from strings. I suggest using `_` instead. We would need to keep the existing logic as a fallback though, if the string does not contain the new delimiter, as we need to still be able to read old files that use the old format.

However, I think it's more important to figure out why we're getting negative ROIs in the first place. That ought to be impossible.

nicost, 2016-07-14:
The negative ROI is generated in the Andor SDK 3 adapter code:
```
   //Adjust for binning
   int binning = GetBinning();
   x = static_cast<unsigned>(aoi_property_->GetLeftOffset() / binning);
   y = static_cast<unsigned>(aoi_property_->GetTopOffset() / binning);

   //Micro-Manager image dims are zero based
   x -= 1;
   y -= 1;
```
It would be easy to assure there that x and y are always positive, but I do not know how that plays out with how the ROI is set on the device (i.e. setting the ROI that is received form the camera should always result in the same ROI, or strange things will happen).  

Either way, progress on my project is now hindered by this bug and I would like to fix it so that I can continue with what I want to work on!

nicost, 2016-07-14:
As to using "_" as a delimiter:  The first place that would need to be changed is in the ```getROITag()``` function defined in ```MMCoreJ.i```, so that ROI Strings will be generated with a "_" as separator rather than "-".  ```MDUtils.getROI() converts the String into a rectangle and should first try to split using "_" and if that does not generate an array of size 4 try it with "-".  MDUtils.setROI() should be modified to use "_" as separator rather than "-".  I think that would indeed work and keep compatibility with old files, but I am not absolutely sure.  In any case, things are not working correctly right now;(

Do you want me to generate a pull request for those changes?

nicost, 2016-07-14:
Pull Request generated.

marktsuchida, 2016-07-15:
In the future, please refer to the pull request by number (or refer to the issue from the pull request) so that an entire discussion doesn't get missed :)
In this case it was #341 (and its replacement #346).

nicost, 2016-07-15:
Currently (i.e. after #346 and #340), I am getting the ROI and binning properties reliably from the metadata.  The Zyla produces negative x and y ofssets whenever the camera is set to full frame and binning is anything other than 1 (i.e, 2,3, 4, and 8 x binning). Thanks for the work on this!

closed 2016-07-18 by nicost