micro-manager/micro-manager #343: Processors have no mechanism to provide nice user feedback

nicost, 2016-07-14:
Processing can take a long time and thing can go wrong.  Without feedback, the user is in the dark about what is going on. Processors could create their own windows for user feedback, but that seem very ugly.  It is also questionable if Processors should create blocking dialogs when something goes wrong.  Logging output is not enough as most users are not continuously checking the log.  

It would be nice to have an area where Processors can provide status updates.  That could either be the main MM window, or the On-The-Fly Processor Pipeline window.  In any case, Processors would need access to a function to provide user feedback.

ChrisWeisiger, 2016-07-14:
We actually have a planned solution for this, which has been pushed to my repo under the "alerts" branch:
https://github.com/ChrisWeisiger/micro-manager/tree/alerts

It adds an API that allows parts of the program to show small, undecorated dialogs along the edges of the screen, which automatically get "compacted" if there's too many of them and which can easily be dismissed/expanded by the user.

It hasn't made it into the "main" mm2 branch yet mostly because we haven't finalized the GUI for it; there have been other priorities. But this issue is very much on our radar. We may as well leave this issue open until the alerts branch gets merged in, anyway.

nicost, 2016-07-14:
Nice!  Looking forward to seeing it in action!

ChrisWeisiger, 2016-07-22:
As of #366 being pulled in, this is now available. Simply call `mm.alerts().showTextAlert(message)` to show a one-time alert to the user. If you expect to want to show multiple messages, you can use an `Object` to tie them together: `mm.alerts().showTextAlert(message, object)` will merge alerts that use the same "owner" object. Or you can show a custom alert with `mm.alerts().showAlert(JComponent contents, Object owner)`.

I'd appreciate it if you could give the API a shot and suggest improvements; it's not had much of a test drive yet.

nicost, 2016-07-23:
When running mm.alerts().showTextAlert(message), nothing happens.  Turns out that the function returns an Alert, but does not show it by itself (if that is the intent, shouldn't it be called getTextAlert?).  The calling code can set the alert to be visible, but also needs to handle making it go away again. That makes it a bit difficult to work with.  Moreover, the Alert for me shows up in the top right corner of the screen, and at first I did not even notice it. Having multiple Alerts appear quickly hides a lot of screen estate, which is a valuable resource.

Would it be possible to center the Alert over the active display window and make it disappear whenever it is no longer the top most window?  That would more or less guarantee that the user sees it, and make it painless 

Alternatively, could there be a "notification" area in the main window where these alerts could be shown?  The ImageJ status line works reasonably well for a similar purpose.

ChrisWeisiger, 2016-07-25:
Taking things in turn:

 * The alert should automatically be made visible by calling `showTextAlert`; you shouldn't need to call `setVisible` yourself. Could you please share the code you're running?

 * Alerts are expected to be dismissed by the user. If you mouse over them, you can either click them to dismiss them (for single-shot alerts) or a close button should show up (for multi-use alerts). This GUI may need some iteration, but it shouldn't be the code's problem to dismiss alerts in any case, unless they _want_ to dismiss the alert for some reason. For example, the alert that shows the number of images collected / remaining during an MDA automatically dismisses itself 5 seconds after the acquisition ends, if it is not dismissed earlier.

 * Large numbers of alerts showing up should be a relatively rare circumstance. I would expect code that wants to use alerts to usually use the mode that combines multiple lines of text into a single dialog (the version with an `owner` parameter), so you'd only get lots of alerts if you had lots of different pieces of code creating alerts. In any case they are easy to dismiss.

 * I'm reluctant to make the alert system try to be too clever about where on the screen it shows alerts, as well as having rules for automatically dismissing them. Alerts shouldn't be _intrusive_, just _noticeable_. Moreover, I don't want to too strongly try to anticipate how they are used or make it too easy to accidentally dismiss them. On the flipside, of course, it should not be too much work to get rid of them either, and if they're too "well hidden" then they don't serve any use.

 * I've never liked the ImageJ status line as an alert mechanism; it's too easy to miss and/or get overwritten by a later alert. However, perhaps we should consider having a specific window dedicated to alerts, so you can hide all alerts by closing the window? Or instead of closing, the window could be "minimized" and an icon is shown somewhere (perhaps in the main window) if there are extant alerts.

nicost, 2016-07-25:
- Visibility: Indeed, they were there, I simply did not notice them (not a good sign for an alert!).  Placement is a real problem, especially with an Inspector window that may hide them.

- Dismissing: There is nothing to indicate to the user that they can dismiss these things.  There is no close button or anything similar.  I now notice that clicking on them dismisses them, but that is certainly not a standard, expected behavior.  in fact, we use a similar thing to indicate loading the hardware, and that one is not dismissable (as far as I know, probably never tried).

- The ShadingProcessor needs to alert the user that it could not execute its function due to whatever reason.  A situation in which the user needs to be alerted may arise with each and every image in live mode.  Is it now the processors responsibility to limit the number of alerts (all of which are the same)?  That is a substantial burden for the processor.  It would be much easier/nice to have a single alert that updates its text, rather than spew new ones.  Or maybe the api code should compare the text and increase a counter when the same text is encountered again rather than show a new alert?
 
- I agree in the downside of the ImageJ status line, but once the user knows it is there, it actually functions relatively well and takes up little space.

ChrisWeisiger, 2016-07-25:
On Mon, Jul 25, 2016 at 8:45 AM, Nico Stuurman <notifications@github.com>
wrote:

>
>    -
>
>    Visibility: Indeed, they were there, I simply did not notice them (not
>    a good sign for an alert!). Placement is a real problem, especially with an
>    Inspector window that may hide them.
>
>
Okay, so at least the code is functioning as it was designed to. :)
Notwithstanding that the GUI needs some improvement.

> -
>
>    Dismissing: There is nothing to indicate to the user that they can
>    dismiss these things. There is no close button or anything similar. I now
>    notice that clicking on them dismisses them, [...]
I've added the close button behavior to all alerts (previously it was only
used for multi-line or custom alerts), so it should be more apparent how
you get rid of them.

> -
>
>    The ShadingProcessor needs to alert the user that it could not execute
>    its function due to whatever reason. A situation in which the user needs to
>    be alerted may arise with each and [...]
You can create a fully-custom alert with a `JLabel` whose text you update
yourself (using the `showAlert` method). Failing that, you should at least
use the `showTextAlert` method that takes an `owner` parameter so that the
alerts get combined into the same dialog. I've updated the documentation to
discourage the non-`owner`-using API method in situations where you may
expect to show multiple alerts.

I'm not yet convinced that the alerts API should be smart enough to
automatically "compress" identical alerts.

nicost, 2016-07-26:
The following use of the API is more or less workable (from the plugin's perspective):

```
JLabel userFeedback= "";
Alert alert = null;

msg = "Something is just not right";
userFeedback_.setText(msg); 
if (alert_ == null) {
     alert_ = studio_.alerts().showAlert(userFeedback_, this);
}
alert_.setVisible(true);
```

Which seems quite complex.

Placement of the alert is still a problem, even now that I know where to expect it, it is still very surprising.  It also happens to appear under the place I usually put the Inspector window (which renders the alert invisible).  I.m.h.o there should be an association with a Micro-Manager window, or it will even be confusing to realize which application this thing is coming from.

Also, the red X icon shows only partially on Windows (its bottom is cut off).

ChrisWeisiger, 2016-07-26:
It should be enough to simply do this:

    JLabel userFeedback_ = new JLabel("Something is not right");
    studio_.alerts().showAlert(userFeedback_, this);

What do you think of the suggestion to have an alerts window, which contains all alerts, along with an icon in the main window that indicates if there are any unseen alerts?

nicost, 2016-07-26:
On 7/26/2016 8:00 AM, ChrisWeisiger wrote:
>
> It should be enough to simply do this:
>
> |JLabel userFeedback_ = new JLabel("Something is not right"); 
> studio_.alerts().showAlert(userFeedback_, this);|

That was my first try.  It leads to an exception the second time itis 
called, putting the burden on the caller to keep track of whether or not 
it already "owns" an alert.  The exception is a lot worse than the issue 
the code tries to report.

||
>
> What do you think of the suggestion to have an alerts window, which 
> contains all alerts, along with an icon in the main window that 
> indicates if there are any unseen alerts?
>

And clicking on it opens the alert window?  Sounds worth a try. Still 
needs quite a bit of thought on how to dismiss alerts, deal with 
repeating alerts as in the example above (which will occur very often, 
i.e. every processor that encounters a bad situation will encounter that 
same situation every time it is called), etc..

ChrisWeisiger, 2016-07-28:
The GUI and API have now been substantially updated. Could you please take a look at the new version? You now no longer need to store your own `JLabel` if you need to update an alert's text multiple times; just keep the `Alert` around and call its `setText(String text)` method. Though in most cases I would expect it to be preferable to use `AlertManager.showCombiningTextAlert(String title, String text)` instead, as that maintains the message history.

nicost, 2016-07-28:
I like this a lot better!  Still a couple of things:
- Can the font of the Message line in the main window be made a bit smaller and similar to the other fonts being used?
- The "clear" and "Clear All" actions do not clear the line in the main window
- After "Clear", the next call to alert.setText(msg) has no effect.  That seems a bug to me (otherwise, there is no way to inform the user of a new problem).
- I do not understand why you so much prefer ```showCombiningTextAlert```.  Very often, the same alert will appear over and over again (for instance when running live mode with an incompatible processor).  For those cases (which I expect to be a majority of using alerts), it is most pleasant to display only the last error/alert, not hundreds of them.
- I still need to wrap the call to show the alert:
```
   private void showAlert(String msg) {
      if (alert_ == null) {
         alert_ = studio_.alerts().showTextAlert(MultiChannelShading.MENUNAME, msg);
      } else {
         alert_.setText(msg);
      }
   }
```
It would be nice if that was not necessary and the api offered a single call to show an alert that can be re-used.

ChrisWeisiger, 2016-07-28:
I can't reproduce the "Clear All" button not clearing the line in the main window. It is however possible to clear the alert that the main window is currently showing, and the main window will still show that alert. I've pushed a change that will clear the message text from the main window after the corresponding alert is cleared. It won't "backtrack" to the next-most-recent alert because that would require some form of history tracking, which I didn't want to get into.

I tried shrinking the font size, but I personally think it looks worse, honestly. This is the one area in the main window where text may change frequently, so IMO it should be easy to read in preference to it being consistent with the other (far more static) labels. But I'll defer to Mark's opinion on this.

As for the API design, you're trying to use it in a way counter to how it was designed to work, which is why it's not working very well.

The design behind the `Alert.setText()` method is specifically for situations where you want to give the user ongoing informative information about some topic. The alert used during MDA to tell the user how many images have been received is a good example of this use. If the user dismisses the alert, they are signalling that they do not care to be informed about that topic, which is why calling `Alert.setText()` does not render the alert visible again. The documentation for the method specifically calls out that it is not appropriate to use the method in most situations.

Combining text alerts are specifically for situations in which multiple errors (or messages) all relating to the same topic may occur. That sounds like precisely the situation you're in. What's wrong with using a combining text alert? If you're experiencing multiple errors, then you _should_ be messaging each one individually. If the user dismisses the alert, it will automatically recur the next time an error happens; if they find the alert irritating, then they can mute it.

In short, your code should simply be calling
    studio_.alerts().showCombiningTextAlert(MultiChannelShading.MENUNAME, msg)
each time it needs to show an alert. No need to track the `Alert` object.

nicost, 2016-07-28:
Imagine running a live acquisition, going through a processor, which encounters a bad situation.  They only way the processor can bring that to the attention of the user is an alert.  The same bad situation occurs up to a 100 times a second.  If I understand the api correctly, calling ```showCombingTextAlert``` will result in 100 copies of the same alert.  That is the last thing I would want to bother the user with.  One of these identical alerts is enough.  On the other hand, even when the user clicks the alert away, the "bad" situation (or possible another bad situation) should still reach the user.  It is really quite detrimental that the user gets no alerts whatsoever anymore (in this case from the processor) after it clicks away a single alert.

So, yes, maybe I am using the api counter to how it is supposed to work, but I do not see much use of it if it does not support this very common usage scenario.

ChrisWeisiger, 2016-07-28:
I spoke with Mark about these issues.

 * The font in the main window has been reduced to match the other text in that window.
 * The max size of the combining text alert has been limited to 100px, which looks a lot better than the old 300px.

We're figuring out some alternative approaches that should make sorting/categorizing alerts easier. We don't want there to be only one alert for a repeating error, but on the other hand the user doesn't necessarily need to see each individual error as a separate line of text, either. Coming up with a good solution will take some time, though.

marktsuchida, 2016-07-28:
Here is roughly what Chris and I sketched out (plus a few things I thought of later):
- Replace `showTextAlert()` with `postProgressAlert()`, with more-or-less the same behavior.
- Replace `showCombiningTextAlert()` with `postEventAlert()` (suggestions for better name appreciated), which will address the case of many recurring warnings/errors (but can also be used for rare warnings/errors).
- `showCustomAlert()` will just be renamed to `postCustomAlert()` and will continue to behave like a progress alert.

(Aside: I suggest the `show` -> `post` rename because whether the alert actually gets "shown" is a GUI detail, not something the calling code should depend on.)

`showEventAlert()` will have the parameters _title_, _channel_, and _text_. The _title_ and _text_ will have more-or-less the same meaning as the current implementation. The _channel_ will be a `Class<?>` and can be `null`.

When displaying, all alerts will be grouped under their title. All progress (and custom) alerts will be visible to the user. All event alerts with a `null` channel will also be visible to the user. When there are multiple alerts under the same title, they can go in a scroll pane, as they currently do (though I think it might be simpler to just display them all, limiting scroll bars to the single one for the whole list -- this should not be a problem if we do a good job at "summarizing" recurrent alerts; there could be an Inspector-like way to collapse a given title, though).

Finally, the channel of an event alert can be set to a particular class (e.g. an empty tag class created for the purpose, or another appropriate class). This should be done for any alert that is expected to repeat itself (and a different class should be used for each distinct warning/error). In this case, the behavior will be
- If no (undismissed) alert with equal title and channel exists, then a new table row shows up in the Messages window, under that title
- If an alert with equal title and channel does exist, a new row _replaces_ any such existing row. The replaced row is not dismissed, but hidden. The table row displays (in small print) the count of undismissed alerts in the same channel, and provides a small button to display the full history (TBD: expand inline or open a separate window?).

In other words, alerts on the same channel (and title) are "summarized" into a single table row when presenting to the user, and only the most recent message text is displayed unless the user chooses to dig down further.

Note that channels are a concept that is hidden from the user; from the user's perspective, Micro-Manager is smartly summarizing multiple occurrences of the "same" alert.

Using a tag class (rather than some one-time object) as the channel means that the "same" alerts from multiple instances of the alert-posting code (e.g. data processor) can easily share the channel. It also obviates the need to hold on to an object and check it for `null` (as in @nicost's `showAlert()` example above).

Other thoughts:
- "Muting" of alerts by the user will still be on a per-title basis
- Alerts sharing a channel but having a different title behave as if they are on different channels
- Make event alerts immutable? Given this design, changing the text of an event alert after posting it is a bad idea. Perhaps make `Alert` immutable, but return a `MutableAlert` (or `UpdatableAlert` or `ProgressAlert`) when posting a progress or custom alert
- How frequently can event alerts be posted to a single channel without running into performance issues when the Messages window is open? This should be characterized (and should be above, say, 200 Hz).
- (lower prio) We could add an overloaded `postCustomAlert()` method allowing to set the text upon creation.
- (lower prio) It might be nice to have a standard version of a progress alert that contains a progress bar, as that seems like a common use case

ChrisWeisiger, 2016-07-29:
This has now been implemented. `AlertManager` now has the following methods:

 * `postAlert(String title, Class<?> group, String text)` for when you anticipate showing multiple messages to the user. Only the most recent message for each title-group pair will be shown, unless the user chooses to review the message history. If the alert is dismissed and this method is invoked, then a new alert will be created. Recommended for the vast majority of situations.

 * `postUpdatableAlert(String title, String text)` for showing status information to the user. Can have its text contents updated via the API (returns an `UpdatableAlert` which exposes `setText()`, while the normal `Alert` interface does not). If the user dismisses it then further calls to `setText()` have no effect.

 * `postCustomAlert(String title, JComponent contents)` works as it did previously. Returns an `UpdatableAlert` solely so that the alert can manipulate the text in the main window if desired.

nicost, 2016-08-10:
Works great!  Only annoyance is that the Messages window does not remember its position between program invocations and always returns to the top left corner of the screen.  Can its position (and possibly size) be remembered in the profile?

marktsuchida, 2016-08-23:
Closing since @nicost fixed the window position issue.

closed 2016-08-23 by marktsuchida