micro-manager/micro-manager #345: mm2: move general-purpose utils files into separate jars

ChrisWeisiger, 2016-07-14:
Now that we have the "libraries" section of the source code, several files that are currently in `org.micromanager.internal.utils` should be moved into their own jars. Off the top of my head:

* DaytimeNighttime (depends on `CompatibilityInterface` for access to constants, which should be moved into DaytimeNighttime directly)
* DirectBuffers (utility as an external jar is questionable, but it is fully standalone)
* FileDialogs (depends on `CompatibilityInterface` for access to same constants DaytimeNighttime uses)
* GUIUtils
* ImageUtils (depends on mmcorej to get image dimensions)
* JavaUtils
* MMDialog
* MMFrame
* MathFunctions
* MemoryUtils
* NumberUtils
* TextUtils
* VersionUtils

Whether each should get its own jar, we try to group related modules into the same jar, or we just toss everything into a single jar is up for debate, as is whether or not they should stay in the `internal` package space and how much tidying-up they need before they can be released as general "APIs". But a lot of these (especially ImageUtils, FileUtils, MMFrame, and MMDialog) are important functionality that we don't want our plugins to have to duplicate.

marktsuchida, 2016-07-15:
I'm not sure I understand why `internal.utils` in particular should go in separate JARs (I feel that separating out Datastores and File I/O, for example, into an independent JAR is higher priority).

For `utils`, presumably it is more important to first develop APIs for some of these so that they can be moved out of `internal` whether or not they stay in the MMStudio JAR or move out of it.

ChrisWeisiger, 2016-07-15:
It's not necessarily that these are higher priority (I created this issue mostly so I wouldn't forget); it's more that they're easier to do. But we do need to decide ''whether'' any one of these ought to be moved. I'd guess that MMFrame and MMDialog are the biggest contenders at the moment since basically any plugin that creates a GUI wants access to them. They do depend on having access to the profile, though.

So I guess, yeah, the issue is not so much what jar they're in as it is that they're currently in `internal`.

Getting the data and display code into separate jars is a pretty clear potential win, but also more work.

marktsuchida, 2020-08-28:
Superseded by #758.

closed 2020-08-28 by marktsuchida