micro-manager/micro-manager #351: Histograms controls do not always act on image in Grayscale mode

nicost, 2016-07-18:
Take an MDA with 3 channels.  Uncheck 'Autostretch' in the Inspector Window and set the Color Mode to "Grayscale" (you may need to take another MDA with these settings).  Move the top triangle in the upper histogram (with the channel slider in the viewer set to channel 1).  Note that the brightness/contrast of the display does not change.  Switch to channel 2 and adjust the brightness/contrast (not that it does not change the display).  Switch to channel 3 and note that brightness/contrast changes as expected.  Switch back to channel 2 and discover that the brightness/contrast sliders of channel 3 act on channel 2.  Switch to channel 1 and see that the brightness/contrast settings of channel 3 act on channel 1.

ChrisWeisiger, 2016-07-18:
This happens in Color mode as well. The problem appears to be that when we try to set the LUT for the `ImageProcessor` (an ImageJ object), we are getting the wrong `ImageProcessor` to operate on (which is readily testable by getting the intensity of a pixel in that processor, and observing that it is the same value regardless of the current active channel). Figuring this out will require figuring out the correct series of dance steps required to get ImageJ to do the right thing, which is always a bit of a fraught process.

ChrisWeisiger, 2016-07-19:
This is now fixed. Thanks for the bug report!

closed 2016-07-19 by ChrisWeisiger