micro-manager/micro-manager #352: Several utility functions were removed from the BeanShell environment in 2.9

nicost, 2016-07-18:
The Micro-Manager documentation for the Script Panel at https://micro-manager.org/wiki/Script_Panel_GUI mentions the following very handy functions:
```
 clear("y"); // Remove method "y" (name in quotes) from the current namespace (i.e., from memory)
  doc(x); // Get javadocs API for class or object x
  inspect(x); // Look at member methods and fields of object x
  update("y"); // Make sure Beanshell uses the most recent definition of method "y"
  update(); // Update all methods in the current namespace
```
These seem no longer to be present in 2.0.  Especially the lack of a ```clear``` function makes it very difficult to work with functions in Beanshell scripts.  I guess these were accidentally removed?  Any chance they could be brought back to life?

ChrisWeisiger, 2016-07-18:
These were not intentionally removed, no. #319 mentions that `inspect` has disappeared and requests its restoration. It's strange that they've vanished, considering how relatively untouched 2.0 is as far as Beanshell is concerned.

nicost, 2016-07-18:
Looking at the 1.4 source code, it looks like the file ```scriptpanel_startup.bsh``` brought in ```inspect() and ```doc()```. ScriptPanel.java in 2.0 has the line: 
```
getResourceAsStream("/org/micromanager/scriptpanel_startup.bsh");
```
but I do not see where that ```scriptpanel_startup.bsh``` lives in the current source code (so I guess that it is no longer loaded on startup).

As for the other functions, the logs of scriptpanel_startup.bsh show:
```
r13821 | mark | 2014-07-15 13:37:30 -0700 (Tue, 15 Jul 2014) | 7 lines
Remove BeanShell update() method

It is not clear why this method was ever added, but it reaches into the
private data structures of BeanShell, and will break if we ever move to
BeanShell2 (which uses modern containers for internal data). The
existence of this method also means that we cannot disallow access to
private fields and methods.
------------------------------------------------------------------------
r13820 | mark | 2014-07-15 13:37:29 -0700 (Tue, 15 Jul 2014) | 4 lines
Remove a pointless BeanShell function

The definition of clear(String) was exactly equivalent to the BeanShell
builtin unset(String).
```
So, I guess this means that we should update the documentation on our website.

ChrisWeisiger, 2016-07-18:
I can answer the first issue: `scriptpanel_startup.bsh` is now located in `mmstudio/src/main/scripts/org/micromanager/scriptpanel_startup.bsh`. You can view it online here:
https://github.com/micro-manager/micro-manager/blob/mm2/mmstudio/src/main/scripts/org/micromanager/scriptpanel_startup.bsh

It may be that the script is not properly getting executed, or that the functions it declares are later getting lost if the "script panel state" is reset.

nicost, 2016-07-18:
Ah, that helps!  To get it loaded when working under Netbeans, I have to add the mmstudio/src/main/scripts directory to the Netbeans project.  Also found an easy way to restore functionality of the ```inspect``` function, which you can find in pull request  #354.  I also updated the documentation on the Wiki (only thing left to do is back-port the fix for the inspect function to 1.4).

closed 2016-07-18 by nicost