micro-manager/micro-manager #361: MMCorePy: crash on access to __dict__

marktsuchida, 2016-07-21:
Egor Zindy reports in http://micro-manager.3463995.n2.nabble.com/Python-crash-bug-report-tp7587318.html:

> here is a reliable way to crash the MMCorePy library:
> 
> ```
> C:\Program Files\Micro-Manager-1.4>python
> Python 2.7.10 (default, May 23 2015, 09:44:00) [MSC v.1500 64 bit (AMD64)] on wi
> n32
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import MMCorePy
> >>> mmc = MMCorePy.CMMCore()
> >>> mmc.getVersionInfo()
> 'MMCore version 8.3.0'
> >>> mmc.__dict__
> {'this':
> C:\Program Files\Micro-Manager-1.4>
> ```

I can confirm this with Python 2.7.12 (x64).

ChrisWeisiger, 2016-07-21:
This appears to be Windows-specific, or at any rate does not reproduce on OSX Python 2.7.5.

marktsuchida, 2020-02-21:
Superseded by pymmcore.

closed 2020-02-21 by marktsuchida